sap-labs-france/ev-server #1511: Charger > Charger registration can fail because of registration with a token

jerome-benoit, 2020-05-18:
So far, happens two times. Need to find a reproducibility pattern. 

Workaround: regenerate the registration token and associated links.

LucasBrazi06, 2020-09-21:
@jerome-benoit any update on this one?

jerome-benoit, 2020-09-21:
> @jerome-benoit any update on this one?

No. So far, the issue happens just one more time. 

The pb is that when it happens, we can't let the charging station at its current state to do proper debugging: the charging station must be supervised ASAP and we have to make it work immediately.

LucasBrazi06, 2020-09-21:
I'll provide an extension of the expiration date for the token as asked by Gerald.
If you cannot reproduce, you can close it.

jerome-benoit, 2020-09-21:
Closed as non reproducible.

closed 2020-09-21 by jerome-benoit