sap-labs-france/ev-server #1556: OCPP parameter update crash when chargepoint is not defined for a charger

jerome-benoit, 2020-06-03:
```
[
      {
            "stack": "Error: OCPP Command 'ChangeConfiguration' has failed
   at Function.handleChargingStationCommand (/home/vcap/app/dist/webpack:/src/server/rest/service/ChargingStationService.ts:1454:13)",
            "previous": {
                  "error": "chargingStation.chargePoints is not iterable",
                  "stack": "TypeError: chargingStation.chargePoints is not iterable
   at WebastoChargingStationVendorIntegration.checkUpdateOfOCPPParams (/home/vcap/app/dist/webpack:/src/integration/charging-station-vendor/ChargingStationVendorIntegration.ts:163:47)
   at Function.handleChargingStationCommand (/home/vcap/app/dist/webpack:/src/server/rest/service/ChargingStationService.ts:1368:43)
   at Function.handleAction (/home/vcap/app/dist/webpack:/src/server/rest/service/ChargingStationService.ts:1097:16)
   at /home/vcap/app/dist/webpack:/src/server/rest/CentralRestServerService.ts:42:13
   at CentralRestServer.restServiceSecured (/home/vcap/app/dist/webpack:/src/server/rest/CentralRestServerService.ts:312:7)",
                  "params": {
                        "key": "MeterValuesSampledData",
                        "value": "Energy.Active.Import.Register"
                  }
            }
      }
]
```

LucasBrazi06, 2020-06-04:
Fixed 9b62b1e50443cfb2dc2707d6c891e19115cfe692

jerome-benoit, 2020-06-04:
The code change looks correct. You want me to validate the fix after deployment?

jerome-benoit, 2020-06-05:
Tested after deployment, it works correctly.

LucasBrazi06, 2020-06-05:
> Tested after deployment, it works correctly.

Thanks Jerome!

closed 2020-06-05 by LucasBrazi06

mytrkr0, 2022-03-31:
Hi 

[image]
It is looking same error but different place.

```js
export default class ChargingStationTemplateBootstrap {
  public static async uploadChargingStationTemplatesFromFile(): Promise<void> {
    // Read File
    Logging.logConsoleDebug("Check point 1");
    let chargingStationTemplates: ChargingStationTemplate[];
   
    try {
      Logging.logConsoleDebug("Check point  2");
      chargingStationTemplates = JSON.parse(fs.readFileSync(`${global.appRoot}/assets/charging-station-templates/charging-stations.json`, 'utf8'));

        try {
          let length_i = chargingStationTemplates.toString();
          Logging.logConsoleDebug(length_i.toString());
        } catch (error) {
          Logging.logConsoleDebug(error.message);
        }
        Logging.logConsoleDebug("Check point  3");
        
    } catch (error) {
      
      await Logging.logActionExceptionMessage(Constants.DEFAULT_TENANT, ServerAction.UPDATE_CHARGING_STATION_TEMPLATES, error);
      
      return;
    }
    // Delete all previous templates
    Logging.logConsoleDebug("Check point  4");
    await ChargingStationStorage.deleteChargingStationTemplates();
    Logging.logConsoleDebug("Check point  5");
    // Update Templates
    for (const chargingStationTemplate of chargingStationTemplates) {
      Logging.logConsoleDebug("Check point  6");
      try {
        // Set the hashes
        Logging.logConsoleDebug("Check point  6-1");
        chargingStationTemplate.hash = Utils.hash(JSON.stringify(chargingStationTemplate));
        chargingStationTemplate.hashTechnical = Utils.hash(JSON.stringify(chargingStationTemplate.technical));
        chargingStationTemplate.hashCapabilities = Utils.hash(JSON.stringify(chargingStationTemplate.capabilities));
        chargingStationTemplate.hashOcppStandard = Utils.hash(JSON.stringify(chargingStationTemplate.ocppStandardParameters));
        chargingStationTemplate.hashOcppVendor = Utils.hash(JSON.stringify(chargingStationTemplate.ocppVendorParameters));
        // Save
        Logging.logConsoleDebug("Check point  7");
        await ChargingStationStorage.saveChargingStationTemplate(chargingStationTemplate);
        Logging.logConsoleDebug("Check point  8");
      } catch (error) {
        Logging.logConsoleDebug("Check point  9");
        error.message = `Charging Station Template ID '${chargingStationTemplate.id}' is not valid: ${error.message as string}`;
        await Logging.logActionExceptionMessage(Constants.DEFAULT_TENANT, ServerAction.UPDATE_CHARGING_STATION_TEMPLATES, error);
        Utils.isDevelopmentEnv() && Logging.logConsoleError(error.message);
        Logging.logConsoleDebug("Check point  10");
      }
    }
  }
}
```
the problem is here:

```js
for (const chargingStationTemplate of chargingStationTemplates) { 

```