sap-labs-france/ev-server #1643: Cars - Converters are not converters.... problem

fcoutellier, 2020-06-30:
Hi Mahdi,

Following the email from Serge I create this task to follow-up on the necessary enhancement to solve it.

[image]

[image]

Check if relevant to remove the converter choice during the car creation for a user and the tab in the visualization of a car in the catalog.

Then, it will then be necessary to ask Florian to verify when creating a few vehicles if the converter matches properly the one in the cars (check with some old cars such as F. Ozon's one or recent ones like Hanno's and ask them directly).

If we have a problem, we will add new fields to enter the power of the converter installed if different from that proposed by EV-Database (phases nbr amps with power display).

fcoutellier, 2020-06-30:
Should be fixed with the following PR:
https://github.com/LucasBrazi06/ev-server/pull/1613
https://github.com/LucasBrazi06/ev-dashboard/pull/1752

This issue is to be moved to icebox till Serge confirms if it works as expected.

jerome-benoit, 2020-06-30:
> Check if relevant to remove the converter choice during the car creation for a user and the tab in the visualization of a car in the catalog.

It's only relevant when there's only one converter for the vehicle. The list choice can stay in grey then. 

++

LucasBrazi06, 2020-06-30:
> > Check if relevant to remove the converter choice during the car creation for a user and the tab in the visualization of a car in the catalog.
> 
> It's only relevant when there's only one converter for the vehicle. The list choice can stay in grey then.
> 
> ++

I already had a meeting with Mahdi and we'll put a drop down to select the converters (standard, option, alternative in EV-DB).

closed 2020-07-09 by fcoutellier