sap-labs-france/ev-server #1679: Back-End - Charging Station in error notification form end-user

fcoutellier, 2020-07-15:
An End-User should be able to push the inforation that a charging station is in error (charging scenario is failing)

A button on the mobile application should be build in a different issue.

jerome-benoit, 2020-07-15:
Related issue: https://github.com/LucasBrazi06/ev-server/issues/1650

LucasBrazi06, 2020-07-15:
> Related issue: #1650

Not related, 1650 is to change a roaming status ;-).

LucasBrazi06, 2020-07-15:
More related to LucasBrazi06/ev-mobile#31 and Epic LucasBrazi06/ev-dashboard#973

jerome-benoit, 2020-07-15:
> > Related issue: #1650
> 
> Not related, 1650 is to change a roaming status ;-).

OCPP 1.6 define a list of charging station statuses. Not need to reinvent the wheel for end user issue report to tenant administrators to link it to one of already predefined statuses at the time of the erro.

LucasBrazi06, 2020-07-15:
> OCPP 1.6 define a list of charging station statuses. Not need to reinvent the wheel for end user issue report to tenant administrators to link it to one of already predefined statuses at the time of the error.

The user just wants to report an issue with the charging station and not necessarily a status error ;-)

jerome-benoit, 2020-07-15:
> The user just wants to report an issue with the charging station and not necessarily a status error ;-)

You add automatically the status at report creation: for example if the status is 'Faulted' since 2 days, the report can be discarded because he tries to use an out of order station.

fcoutellier, 2020-07-16:
What about if the charging station is simply not usable. Physical damage, connector not accessible, vandalize. Such cases will not be necessarily detected/handled by the OCPP status. However, the charging station will actually not be usable any longer.

jerome-benoit, 2020-07-16:
> What about if the charging station is simply not usable. Physical damage, connector not accessible, vandalize. Such cases will not be necessarily detected/handled by the OCPP status. However, the charging station will actually not be usable any longer.

You still add automatically all the targeted charging station information snapshot (including the status) to the report. The report granularity is connector -> charging station.

fcoutellier, 2020-08-27:
To be tested with the UI

closed 2020-08-27 by fcoutellier