sap-labs-france/ev-server #1710: Smart Charging: Handle also two phases cars

jerome-benoit, 2020-07-31:
Most of the Volkswagen electric vehicles have a current transformer that is 2 phase only.

LucasBrazi06, 2020-07-31:
I don't get this, could you elaborate ?

jerome-benoit, 2020-07-31:
> I don't get this, could you elaborate ?

Some electric cars are charging with 2 phases only.

LucasBrazi06, 2020-07-31:
> > I don't get this, could you elaborate ?
> 
> Some electric cars are charging with 2 phases only.

Ok, it's like using two single phase converters to accelerate the charge (2x8 or 2x16...).

jerome-benoit, 2020-07-31:
> > > I don't get this, could you elaborate ?
> > 
> > 
> > Some electric cars are charging with 2 phases only.
> 
> Ok, it's like using two single phase converters to accelerate the charge (2x8 or 2x16...).

Probably, I do know how it's done internally and electrically but Volkswagen seems to use 2 phases converters a lot.

jerome-benoit, 2020-08-09:
At `ChargingSchedulePeriod` creation, its attribute `numberPhases` must be properly filled.

LucasBrazi06, 2020-08-09:
> At `ChargingSchedulePeriod` creation, its attribute `numberPhases` must be properly filled.

It's not mandatory and I don't think this will bring any added value compared to not providing it.

jerome-benoit, 2020-08-09:
If you ask the fair share smart charging algorithm to take into account only one phase for one phase car for defining the charging profile, then you have to specify the number of phases used in the profile, otherwise the charging station will not be able to apply the same limit and the overall max power consumption in a zone will not match the real maximum, which not correct. 
If you have a lot of cars using one phase charging at the same time, the gain over the available power for others is significant, especially in power constrained zones which are very commons.

fcoutellier, 2020-08-27:
@jerome-benoit : Could you please support me with the functional test (real charging)?

jerome-benoit, 2020-08-27:
> @jerome-benoit : Could you please support me with the functional test (real charging)?

That task either depends on the car electric specification integration in the smart charging or on the possibility to define the number of phases at charging profile creation. 
I think the first is less error prone and require less end-user actions. 

For testing, a single or two phases car is needed (Jaguar or Golf), plug it on a charging station that permit to have per phase metervalues, define a charging profile on the charging station for the next hours, start a transaction attached to the car electric specs, inspect the metervalues sent and the charging profile settings on the charging station.

jerome-benoit, 2020-09-17:
From last exchange on that topic, I think the last bits to add to properly handle the number of phases in smart charging is to specify it at the charging profile level.

LucasBrazi06, 2020-09-17:
> From the last exchange on that topic, I think the last bits to add to properly handle the number of phases in smart charging is to specify it at the charging profile level.

Just created an issue to handle this LucasBrazi06/ev-dashboard#1965

closed 2020-10-11 by LucasBrazi06