sap-labs-france/ev-server #1740: Charging Plan > fix connector limit method

OliveGerste, 2020-09-01:
Started in branch https://github.com/LucasBrazi06/ev-server/tree/Fix-getCurrentLimitMethod

jerome-benoit, 2020-10-15:
What is exactly expected to fix in that issue?

OliveGerste, 2020-10-15:
@jerome-benoit 
The issue was about the getCurrentConnectorLimitFromProfiles-method, which is used e.g. to add the limit to the consumption.
The method was not capable of:
- returning a limit, when the connector id is 0
- returning  a limit, when there is only one period in the profile
- returning  a limit, when the profile type is recurring
- returning  a limit, when the last period of the cp is active

jerome-benoit, 2020-10-19:
> @jerome-benoit
> The issue was about the getCurrentConnectorLimitFromProfiles-method, which is used e.g. to add the limit to the consumption.
> The method was not capable of:
>
> * returning a limit, when [...]

Ok, and it's a work in progress or QA related issue?

OliveGerste, 2020-10-20:
@jerome-benoit It is QA related. 
Related PR: https://github.com/LucasBrazi06/ev-server/pull/1765

jerome-benoit, 2020-10-28:
> @jerome-benoit It is QA related.
> Related PR: #1765

That one is a bit time consuming to test on real charging station. 
How have you checked that the value returned is correct during your development?

OliveGerste, 2020-10-28:
@jerome-benoit 
I tested it manually via database modifications.
I added different kind of profiles with different limits to the database and checked if the limit is correct in the debugger.

closed 2021-08-05 by LucasBrazi06