sap-labs-france/ev-server #1749: Optimizer Request > inconsistency between sum of phases and minCurrent/maxCurrent

OliveGerste, 2020-09-07:


jerome-benoit, 2020-10-05:
QA: not tested. 

@OliveGerste: what are the validation tests?

jerome-benoit, 2020-10-28:
> QA: not tested.
> 
> @OliveGerste: what are the validation tests?

@OliveGerste: ping?

OliveGerste, 2020-10-28:
@jerome-benoit 
The parameter min/max current in the optimizer request is related to the canLoadPhase parameters.
E.G. minCurrent must be (canLoadPhase1 + canLoadPhase2 + canLoadPhase3) * minCurrentPerPhase.
This was not the case all the time, which let the request fail sometimes. The reason for that was mixing parameters from the metervalues (car) with parameters from the station. (Phases from car not always equal to phases from station)

So it is kind of hard to test.
I tested it with all different type of stations and sending meterValues manually from postman.

jerome-benoit, 2020-10-28:
> @jerome-benoit
> The parameter min/max current in the optimizer request is related to the canLoadPhase parameters.
> E.G. minCurrent must be (canLoadPhase1 + canLoadPhase2 + canLoadPhase3) * [...]

I've recently added per phase measurand support in ev-simulator. I think it's now pretty easy to add per phase static value measurand to trigger phase imbalance or one or two phase(s) charging scenario with it. 
Which measurand are you using to determine canLoadPhase{1,2,3}?

LucasBrazi06, 2020-10-28:
@jerome-benoit 
Should be done in the Unit Test, not the simulator.
Don't waste your time in this area.
If it cannot be tested in QA, let it go.

WillyLellouch, 2021-01-14:
Could I have an update on this ? Is this tested in unit tests now or do I need to test it with the simulator ?

OliveGerste, 2021-01-15:
@WillyLellouch There are several unit tests for creating requests to the optimizer, which should also cover this bug. (Unfortunately there can always happen unexpected things in reality)
I have never seen the error again in prod, since it this fix.

WillyLellouch, 2021-01-15:
@jerome-benoit @LucasBrazi06 
Is this still relevant to QA testing then ?

WillyLellouch, 2021-02-11:
ping ?

closed 2021-08-05 by LucasBrazi06