sap-labs-france/ev-server #1759: Templates: Connectors handling issue

jerome-benoit, 2020-09-15:
When a charging stations have more connectors than the matching template (new version of the same model for example), the template on the connector fails somehow to be applied on the right matching connector type and no correct information are displayed.    

Example: Schneider-CB1000 on https://testcharger.qa-evse.cfapps.eu10.hana.ondemand.com/, one connector added.  

Detecting the new version do not seems possible on Schneider.

LucasBrazi06, 2020-09-15:
Why do we have a connector K for this charger (which would correspond to connector id 9)?

[image]

I would expect A (connector id 1).

jerome-benoit, 2020-09-15:
> 
> Why do we have a connector K for this charger (which would correspond to connector id 9)?
> 
>  <https://user-images.githubusercontent.com/25266945/93219495-bd6fbf00-f76b-11ea-80ce-f16160274b3e.png>
> I would expect A (connector id 1).
> 
> 

That’s part of the issue: the charging station is sending the right information but we do not handle them correctly.

--
Jérôme Benoit aka fraggle
Piment Noir - https://piment-noir.org
OpenPGP Key ID : 27B535D3
Key fingerprint : B799 BBF6 8EC8 911B B8D7 CDBC C3B1 92C6 27B5 35D3

jerome-benoit, 2020-09-15:
Related bugs to half configured station through templates: 

```
[
      {
            "stack": "Error: OCPP Command 'ChangeConfiguration' has failed
   at Function.handleChargingStationCommand (/home/vcap/app/dist/webpack:/src/server/rest/service/ChargingStationService.ts:1436:13)",
            "previous": {
                  "error": "Cannot read property 'numberOfConnectedPhase' of undefined",
                  "stack": "TypeError: Cannot read property 'numberOfConnectedPhase' of undefined
   at Function.getNumberOfConnectedPhases (/home/vcap/app/dist/webpack:/src/utils/Utils.ts:720:29)
   at SchneiderChargingStationVendorIntegration.convertLimitAmpToAllPhases (/home/vcap/app/dist/webpack:/src/integration/charging-station-vendor/ChargingStationVendorIntegration.ts:557:32)
   at SchneiderChargingStationVendorIntegration.checkUpdateOfOCPPParams (/home/vcap/app/dist/webpack:/src/integration/charging-station-vendor/ChargingStationVendorIntegration.ts:139:44)
   at Function.handleChargingStationCommand (/home/vcap/app/dist/webpack:/src/server/rest/service/ChargingStationService.ts:1350:43)
   at Function.handleAction (/home/vcap/app/dist/webpack:/src/server/rest/service/ChargingStationService.ts:1107:16)
   at /home/vcap/app/dist/webpack:/src/server/rest/CentralRestServerService.ts:42:13
   at CentralRestServer.restServiceSecured (/home/vcap/app/dist/webpack:/src/server/rest/CentralRestServerService.ts:334:7)",
                  "params": {
                        "key": "maxintensitysocket",
                        "value": "32"
                  }
            }
      }
]
```

jerome-benoit, 2020-09-16:
Schneider confirms that the connectorID sent are 11 and 12, which not standard. 
So the issue is only on the template application.

LucasBrazi06, 2020-09-16:
> Schneider confirms that the connectorID sent are 11 and 12, which not standard.
> So the issue is only on the template application.

So they will not consider it as a bug!?

jerome-benoit, 2020-09-16:
> > Schneider confirms that the connectorID sent are 11 and 12, which is not standard.
> > So the issue is only on the template application.
> 
> So they will not consider it as a bug!?

Now, yes but it seems nobody noticed before ;)

LucasBrazi06, 2020-09-16:
> Now, yes but it seems nobody noticed before ;)

Ok lol !!!
I fixed the NPE but I need to check if the code is safe also after.

jerome-benoit, 2020-09-16:
I think that's when charging station match a template but have more connectors than the ones in it, the unmatched connectors should still be manually configurable. 

And maybe add a connector type matching if the connector id matching fails at the template application.

LucasBrazi06, 2020-09-16:
> I think that's when charging station match a template but have more connectors than the ones in it, the unmatched connectors should still be manually configurable.
Yes, but we should consider that this is a bug to be fixed by Schneider then no action on our side.
You can provide a workaround in the template for connector 11 in the meantime for the given firmware :-)

> And maybe add a connector type matching if the connector id matching fails at the template application.
No, too dansgerous.

jerome-benoit, 2020-09-17:
> Yes, but we should consider that this is a bug to be fixed by Schneider then no action on our side.
> You can provide a workaround in the template for connector 11 in the meantime for the given [...]

Well, Schneider have a bug and we have also a bug: we do not handle properly different version of the same charging station model. All vendors are doing such evolution on their products so hardcoded  workarounds is not a viable solution. 
In Linux kernel drivers, that kind of workarounds are handled properly with C macros saying that for example SATA disk made by X with firmware version Y do not handle properly feature Z,...  
And in the driver code paths, feature Y is blacklisted on that hardware. 
That's a the only viable solution when it comes to talk to a large amount of unknown hardware via a standardised protocol.      

> > And maybe add a connector type matching if the connector id matching fails at the template application.
> No, too dangerous.

Vendors like Delta sell modular charging station: same model everywhere with global shared specs for AC/DC, AC (we do not handle AC yet), connectors can all be different.

LucasBrazi06, 2020-09-17:
@jerome-benoit We have to discuss this but if it's a bug at Schneider side we have nothing to do on our side.

LucasBrazi06, 2020-09-17:
@jerome-benoit If it's blocking production, do not provide a template for this model if possible.

jerome-benoit, 2020-09-17:
> @jerome-benoit If it's blocking production, do not provide a template for this model if possible.

It's already the case ;)

For Keba, we do not handle properly: 
+ master/slave setup; 
+ mono phase usage on tri phase charging station. 

And a customer is using them on production with both features. 
So no template definition done on Keba until we can have support in templates for both. 
And in that case, advanced features such as charging profiles are disable. 
Since features support announcement is part of the OCPP protocol, we can enable features on the charging station supported by default and explicitly state that no official support by e-mobility will be provided. 

For Schneider, that's another story: it's a new version of an already existing model with a domestic plug added. The connector numbering is buggy on Schneider but even with the right numbering, the type2 plug will be properly handled but the code is not able to cope with the added plug (no displaying at all on the charging station details, displayed on the charging stations list). 
So there's also bugs on our side.

LucasBrazi06, 2020-09-17:
> It's already the case ;)

Perfect!

> For Keba, we do not handle properly:
> * master/slave setup;

Yes, we need to implement this feature in the templates for slave connectors

> * mono phase usage on tri phase charging station.

For this we cannot know until we have an OCPP property. In that case we can check this property and set the proper number of phases.
We have already the issue on Schneider and we haven't removed it from the template. This requires a manual action until we have the OCPP param.

> Since features support announcement is part of the OCPP protocol, we can enable features on the charging station supported by default and explicitly state that no official support by e-mobility will be provided.

This we didn't want to do, because you don't know if it has been tested by us or not and if it is working with our backend.

> For Schneider, that's another story: it's a new version of an already existing model with a domestic plug added. The connector numbering is buggy on Schneider but even with the right numbering, the [...]

Yes we have to fix it then!
Did you already create issues?
3 issues: Slave connector in template, Handling of nbr connected phases with OCPP params, Fix of Schneider connectors in UI?

jerome-benoit, 2020-09-17:
> For this we cannot know until we have an OCPP property. In that case we can check this property and set the proper number of phases.
> We have already the issue on Schneider and we haven't removed it [...]

The GitHub issue expect to be able to manually do the setup and adjust the calculation, no OCPP parameters exists for all vendors. All the calculation will be simplified and optimised if we use amperage per phase as a specification reference. 
Full rationale on that topic available here: https://wiki.piment-noir.org/doku.php/en:cs:modelling_multi-phased_electrical_system_interconnexion  

> This we didn't want to do, because you don't know if it has been tested by us or not and if it is working with our backend.

We do not care, we do not support it explicitly: out of template charging station usage with user agreement approval before enablement. Just like OS vendors do not officially support all USB plugs, all SATA disks, all production chain robot hardware, etc. in the world but does not forbid end-user to use the one they want, especially if they have paid for the OS ;)  

> Yes we have to fix it then!
> Did you already create issues?
> 3 issues: Slave connector in template,

Already exist. 

> Handling of nbr connected phases with OCPP params

Already exist. 

> Fix of Schneider connectors in UI?

Does not exist, it's more generic than that, it's about how to cope with hardware evolution and/or modular hardware in templates for the same charging station model. That needs more thoughts before being transcribed in GitHub issues.

LucasBrazi06, 2020-09-17:
> The GitHub issue expect to be able to manually do the setup and adjust the calculation, no OCPP parameters exists for all vendors. All the calculation will be simplified and optimised if we use [...]

Keep in mind that template's goal is to avoid manual setup.
If you allow to change props in the charging station, it'll be overridden next time the template will be changed and you will have to reapply the manual change again (charge@home use case).
This will require a complex dev to handle overridden props set by user to never be changed again + a way to force to reapply the template in case of incompatible changes.
We must find a simple way to turn 3 phases charger to 1 phase **automatically** as this does not scale.

And I don't see what the amp per phase will solve here if we don't know if it's 1 or 3 phases wired charger!

> We do not care, we do not officially support it explicitly: out of template charging station usage with user agreement approval before enablement. Just like OS vendors do not officially support all USB plugs, all SATA disks, all production chain robot hardware, etc. in the world but does not forbid end-user to use the one they want, especially if they have paid for the OS ;)

We are not Microsoft and we agree with Gerald to deliver in the template only Charging Station Supported AND Tested with our backend by us. Nothing to discuss here and you know it already.

> it's about how to cope with hardware evolution and/or modular hardware in templates for the same charging station model. That needs more thoughts before being transcribed in GitHub issues.

We considered the hardware not to change without a change in the model or any other standard mandatory props we have in OCPP. If this is the case, we have indeed to find a solution, give me a charger which has this problem?

jerome-benoit, 2020-09-17:
Meeting minutes for outside readers: 

+ To cope with model and vendor changes of the same hardware and firmware: allow to factorise configuration with an inheritance. Elegant and easiest solution to implement. 
+ To cope with hardware changes matching the same model: at the connector level, there's nothing to do: if there's no connectorID matching the connectorID in the template, nothing will be applied from it.
+ To deal with multiples values choices in templates (example use case: number of phases): It's indeed a complex topic that will need a lengthy development to be properly added. A workaround could be implemented by the next item.    
+ To deal with fully featured charging station usages outside the templates: allow templates system opt-out at a charging station level with an admin user approved agreement on 'configure and use at your own risk, no responsibilities  or support from e-mobility'. Once outside the templates system, all configuration fields are manually configurable including advanced features enablement. Some features enablement can be done automatically by parsing the OCPP key 'SupportedFeatureProfiles' to ease user experience. 
+ To deal with modular charging stations: wait for more real word experience before designing it.
+ To have charging station electrical specification using amperage per phase at a low level (user experience remains the same): evaluate the development impact, pros and cons, etc.  

I do not think I've missed anything. But if have, it's easy to fix ;)

LucasBrazi06, 2020-09-17:
> I do not think I've missed anything. But if have, it's easy to fix ;)

This is a perfect summary of what we discussed!
Thanks for the minutes!

closed 2021-08-05 by LucasBrazi06