sap-labs-france/ev-server #1760: Transaction > Store the phases used in transaction

OliveGerste, 2020-09-16:


jerome-benoit, 2020-10-05:
QA: not tested. Basically it's matter of checking a new field in the transactions collection? 

@OliveGerste: have you validated it with simulated charging stations doing transactions?

OliveGerste, 2020-10-06:
@jerome-benoit 
The new property 'phasesUsed', which is set for AC-Ssessions in the transaction collection is used to tell the optimzer, which phases are used from the charging car. It is calculated from the meterValues and only effects something, when a car is single or duo phased.
I tested via postman, if the property is set for the different cases (set with the first meterValue) and if the optimizer request is build correctly.

jerome-benoit, 2020-10-07:
Ok. I've checked that in the database have the new field is filled and that without metervalues received, the new field is not added.  

Same remark for your postman collection, I think it worths the effort to merge it with the one in ev-server repository ;)

Thanks.

closed 2020-10-07 by LucasBrazi06