sap-labs-france/ev-server #1763: Docker fails to build

adyoull, 2020-09-17:
Docker fails to build on clean install of Ubuntu 20.04LTS Server, SSH keys are all correctly set, here is the output, the MongoDB aspect correctly builds, its after this stage it fails as below:

> Please make sure you have the correct access rights
> and the repository exists.
> fatal: clone of 'git@github.com:LucasBrazi06/ev-sap-charging-station-templates.git' into submodule path '/home/andrew/ev-server/src/assets/charging-station-templates' failed
> Failed to clone 'src/assets/charging-station-templates'. Retry scheduled
> Cloning into '/home/andrew/ev-server/src/assets/configs-aws'...
> git@github.com: Permission denied (publickey).
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> fatal: clone of 'git@github.com:LucasBrazi06/ev-aws.git' into submodule path '/home/andrew/ev-server/src/assets/configs-aws' failed
> Failed to clone 'src/assets/configs-aws'. Retry scheduled
> Cloning into '/home/andrew/ev-server/src/assets/configs-ci'...
> git@github.com: Permission denied (publickey).
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> fatal: clone of 'git@github.com:LucasBrazi06/ev-ci.git' into submodule path '/home/andrew/ev-server/src/assets/configs-ci' failed
> Failed to clone 'src/assets/configs-ci'. Retry scheduled
> Cloning into '/home/andrew/ev-server/src/integration/pricing/convergent-charging'...
> git@github.com: Permission denied (publickey).
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> fatal: clone of 'git@github.com:LucasBrazi06/ev-sap-pricing-convergent-charging.git' into submodule path '/home/andrew/ev-server/src/integration/pricing/convergent-charging' failed
> Failed to clone 'src/integration/pricing/convergent-charging'. Retry scheduled
> Cloning into '/home/andrew/ev-server/src/integration/refund/concur'...
> git@github.com: Permission denied (publickey).
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> fatal: clone of 'git@github.com:LucasBrazi06/ev-sap-refund-concur.git' into submodule path '/home/andrew/ev-server/src/integration/refund/concur' failed
> Failed to clone 'src/integration/refund/concur'. Retry scheduled
> Cloning into '/home/andrew/ev-server/src/integration/smart-charging/sap-smart-charging'...
> git@github.com: Permission denied (publickey).
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> fatal: clone of 'git@github.com:LucasBrazi06/ev-sap-smart-charging.git' into submodule path '/home/andrew/ev-server/src/integration/smart-charging/sap-smart-charging' failed
> Failed to clone 'src/integration/smart-charging/sap-smart-charging'. Retry scheduled
> Cloning into '/home/andrew/ev-server/src/assets/charging-station-templates'...
> git@github.com: Permission denied (publickey).
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> fatal: clone of 'git@github.com:LucasBrazi06/ev-sap-charging-station-templates.git' into submodule path '/home/andrew/ev-server/src/assets/charging-station-templates' failed
> Failed to clone 'src/assets/charging-station-templates' a second time, aborting
> make: *** [Makefile:20: submodule-update] Error 1

jerome-benoit, 2020-09-17:
Hello, 

Could you please give us the commands use generate the docker image? 
(Always give all the steps you've follow when it comes to bug report, and also the error outputed)  

You do not need to have access to the submodules for the docker image to be run. 
So you can add - before the culprit command in the Makefile target to allow make to continue despite the error. 

And then make a pull request once the fix is successfully tested.

jerome-benoit, 2020-09-17:
In fact, the Makefile already handle that case, just run: 

make SUBMODULES_INIT=false

adyoull, 2020-09-17:
> Hello,
>
> Could you please give us the commands use generate the docker image?
> (Always give all the steps you've follow when it comes to bug report, and also the error outputed)
>
> You do not need to [...]

Hi Jerome,

I literally rang "Sudo Make" to begin the docker process from within the docker sub-directory. using your answer "make SUBMODULES_INIT=false" I've managed to successfully build the docker image.

Thanks for your help.

closed 2020-09-17 by adyoull