sap-labs-france/ev-server #1769: Charging Station Template > handle phaseAssignment, when charging station is registered

OliveGerste, 2020-09-23:
The property phaseAssignment (to grid) needs to be added in the chargingStationTemplate and has to be initialized with a default value, when a siteArea is assigned in the token.
The issue is created from the requested changes in https://github.com/LucasBrazi06/ev-server/pull/1754
@jerome-benoit Serge told me to come back to you for this. Feel free to reach out to me, for clarification.

jerome-benoit, 2020-09-23:
Hello,
> The property phaseAssignment (to grid) needs to be added in the chargingStationTemplate and has to be initialized with a default value, when a siteArea is assigned in the token.
> The issue is created [...]

I think it’s not necessary to add a new tunable in the template because it’s not a charging station dependent specification, it’s installation dependent.
By default, assume RST phase rotation on all charging stations.

--
Jérôme Benoit aka fraggle
Piment Noir - https://piment-noir.org
OpenPGP Key ID : 27B535D3
Key fingerprint : B799 BBF6 8EC8 911B B8D7 CDBC C3B1 92C6 27B5 35D3

LucasBrazi06, 2020-09-23:
> I think it’s not necessary to add a new tunable in the template because it’s not a charging station dependent specification, it’s installation dependent. By default, assume RST phase rotation on all charging stations.

I would like that all tech default values are also in the template even if it's not charger dependant.

jerome-benoit, 2020-09-23:
> I think it’s not necessary to add a new tunable in the template because it’s not a charging station dependent specification, it’s installation dependent. By default, assume RST phase rotation on all [...]
Since the number of phasse is already in, that’s fine.
So there’s a tunable at the connector level ?
What’s it name ? And which values can it take ?

--
Jérôme Benoit aka fraggle
Piment Noir - https://piment-noir.org
OpenPGP Key ID : 27B535D3
Key fingerprint : B799 BBF6 8EC8 911B B8D7 CDBC C3B1 92C6 27B5 35D3

OliveGerste, 2020-09-23:
Hello,

It is dependent on the number of phases of the connector, but I/we can also modify the enrichChargingStationConnectorWithTemplate method and set the property, when a Chargingstation is registered and a siteArea is assigned. 
So we can prevent the property from being set to charging stations without a template.
@LucasBrazi06 What do you think?

jerome-benoit, 2020-09-23:
> It is dependent on the number of phases of the connector, but I/we can also modify the enrichChargingStationConnectorWithTemplate method and set the property, when a Chargingstation is registered and a siteArea is assigned.

At the template level, we do not have yet semantic checks. So the person in charge will have to know what he is doing when defining the default phase rotation.  

* Tri-phased: 
  * tunable name shall be defaultPhaseRotationAssignment; 
  * values shall be 0, 120, 240
* Mono-phased: 
  * idem tunable name
  * values shall be 1, 2 ,3 

> So we can prevent the property from being set to charging stations without a template.

I do think that's the plan: in the templates you only set the default phase rotation/assignment value. 
Then admins are always free to change that value in any case.

OliveGerste, 2020-09-24:
Sounds logical. 
The data model of the property looks like this:
export interface PhaseAssignmentToGrid {
  csPhaseL1: OCPPPhase.L1 | OCPPPhase.L2 | OCPPPhase.L3;
  csPhaseL2: OCPPPhase.L1 | OCPPPhase.L2 | OCPPPhase.L3;
  csPhaseL3: OCPPPhase.L1 | OCPPPhase.L2 | OCPPPhase.L3;
}
Should I take care of the implementation?

LucasBrazi06, 2020-09-24:
> Sounds logical.
> The data model of the property looks like this:
> export interface PhaseAssignmentToGrid {
> csPhaseL1: OCPPPhase.L1 | OCPPPhase.L2 | OCPPPhase.L3;
> csPhaseL2: OCPPPhase.L1 | OCPPPhase.L2 [...]

Do you feel confident to enhance the template and change the code when it's applied? ;-)
You can request Jerome's assistance.

OliveGerste, 2020-09-24:
If Jerome is available, this would be great.

jerome-benoit, 2020-09-24:
> If Jerome is available, this would be great.

Not before the next week ...

LucasBrazi06, 2020-09-24:
> > If Jerome is available, this would be great.
> 
> Not before the next week ...

I will assist you then, we can have a meeting this afternoon.

jerome-benoit, 2020-10-12:
Seems to work. 

I think that supporting the OCPP standard parameter ConnectorPhaseRotation to configure the rotation would be a must. The key format is a comma separated list: connectorID.RST, connectorID.STR, etc.  

If the key does not exist or is empty, fallback to default RST
Else assign the phase rotation in the key to the connector ID

closed 2021-08-05 by LucasBrazi06