sap-labs-france/ev-server #1772: Smart Charging - Refactor build car method and handle phases properly

OliveGerste, 2020-09-25:


jerome-benoit, 2020-10-05:
QA: not tested.

@OliveGerste: what are the validation tests?

OliveGerste, 2020-10-06:
@jerome-benoit  
I tested via postman, that all the cases for triggering the smart charging still work correctly:
Test cases I handled:
Single phased station (with single phased car, three phased car) (TransactionStart, StatusChange, first meterValue with phases used)
Three phased station (with single phased car, three phased car) (TransactionStart, StatusChange, first meterValue with phases used)
Dc Charger (TransactionStart, StatusChange, first meterValue)

It is a bit cumbersome to do it with postman. 
I think it is also okay to watch the logs for triggering the smart charging, because we have all these cases in production

jerome-benoit, 2020-10-07:
Ok, so I think that issue can be closed and that you probably need to merge your updated postman collection into the ev-server one. 

Thanks.

closed 2020-10-07 by LucasBrazi06