sap-labs-france/ev-server #1805: Unit tests: Failure at createContext and run because of user session hash change

jerome-benoit, 2020-10-03:
A HTTP error 403 is returned at createContext and during certain test like the settings related one. 
If I comment out that code 

 ```
    if (global.userHashMapIDs.has(`${tenantID}#${userID}`) &&
        global.userHashMapIDs.get(`${tenantID}#${userID}`) !== userHashID) {
        throw new AppError({
          source: Constants.CENTRAL_SERVER,
          errorCode: StatusCodes.FORBIDDEN,
          message: 'User has been updated and will be logged off',
          module: MODULE_NAME,
          method: 'isSessionHashUpdated',
          user: req.user
        });
      }
```

in isSessionHashUpdated method, the error disappears.

LucasBrazi06, 2020-10-03:
@mahdibenromdhane 
Double check if this is related to the unit test that reassigns the badge from one user to another one.
This is the only scenario that could call this method.

LucasBrazi06, 2020-10-05:
This property in the config.json must be set to false when running the unit tests:
"monitorDBChange": false,

closed 2020-10-05 by LucasBrazi06