sap-labs-france/ev-server #1832: Tags management > Tag with transactions cannot be attributed to a user

jerome-benoit, 2020-10-15:
It can also not be deleted.

LucasBrazi06, 2020-10-15:
Normal behavior. Not an issue.

LucasBrazi06, 2020-10-15:
@mahdibenromdhane 
We need just to check only the sessions with userID is provided (not null).
So just refine the MongoDB request if it's not done yet.

jerome-benoit, 2020-10-15:
> Normal behavior. Not an issue.

Tag in badges in companies are reused. So tags must be able to change owner.

LucasBrazi06, 2020-10-15:
> Tag in badges in companies are reused. So tags must be able to change owner.

This is not a use case we support today neither with SAP badges nor with end-user badges, they are all nominative and then cannot be reassigned.

mahdibenromdhane, 2020-10-27:
Tags With only anonymous sessions (userID null) can be:
- Reassigned to another user and a popup should show to ask if we want to assign the orphan sessions to the new user: 
[image]

- Deleted

- If deleted and then create a tag with the same ID a popup should show to ask if we want to assign the orphan sessions to the user assigned to the new tag. 
[image]

WillyLellouch, 2021-01-13:
@mahdibenromdhane @jerome-benoit 

The functionality works as described in different contexts.
I had to go in the database to create Tags without users and sessions with those tags. Is it even possible to start sessions without a user ? When I tried to do so, it return an error, which I think is correct behaviour.

LucasBrazi06, 2021-01-13:
> @mahdibenromdhane @jerome-benoit
> 
> The functionality works as described in different contexts.
> I had to go in the database to create Tags without users and sessions with those tags. Is it even possible to start sessions without a user ? When I tried to do so, it returned an error, which I think is correct behaviour.

Yes you can start a session without a user when the access control is disabled in the Site Area:

[image]

WillyLellouch, 2021-01-13:
Ok I tried it that way and the functionality works as intended here too

LucasBrazi06, 2021-01-13:
Ok, then I'll close the issue.
Thanks!

closed 2021-01-13 by LucasBrazi06