sap-labs-france/ev-server #1837: Charging Profiles - convertFromVendorChargingSchedule does not work for profiles pushed in watts

OliveGerste, 2020-10-16:
[image]

Limit for the composite schedule is returned in amps from the station. 
The value is divided by the voltage, because the server thinks it is returned in watts.
(appearing on dc station)

LucasBrazi06, 2020-10-16:
SAP-Mougins-15 uses Watt in the template and we push the charging profile in Watts and the getCompositeSchedule() returns Amps !? WTF LOL !!!

[image]

We must align the charger config via OCPP params to use the right unit provided in the Template in both setting and reading the charging profiles.

@jerome-benoit Could you check?

jerome-benoit, 2020-10-16:
If you are sure that the unit specified in the profile is properly set to Watt, then it's a firmware bug. 
There's no OCCP parameter to set, everything is in the charging profile. The read-only parameter ChargingScheduleAllowedChargingRateUnit have Current,Power value. 

The current firmware version should support correctly A as a unit, so the workaround is to switch to A in the template.

LucasBrazi06, 2020-10-16:
> If you are sure that the unit specified in the profile is properly set to Watt, then it's a firmware bug.
> There's no OCCP parameter to set, everything is in the charging profile. The read-only [...]

Yes I'm sure it is set in Watt because this charger supported only charging profile in Watt at the time we set it up and we never changed that.

With new firmware updates according this OCPP param, they now support Amps:

[image]

And I think we should rely on the OCPP param to check which unit to use, and then we can remove it from the template:

[image]

When we request the profile, do we use the param chargingRateUnit in the getCompositeSchedule?

[image]

Let me know!
But there is room for improvement here.

jerome-benoit, 2020-10-16:
The bug is on our side: 

```
[
      {
            "chargingProfile": {
                  "chargingStationID": "SAP-Mougins-15",
                  "connectorID": 2,
                  "chargePointID": 1,
                  "profile": {
                        "chargingProfileId": 2,
                        "chargingProfileKind": "Absolute",
                        "chargingProfilePurpose": "TxProfile",
                        "transactionId": 1272648945,
                        "stackLevel": 2,
                        "chargingSchedule": {
                              "chargingRateUnit": "A",
                              "chargingSchedulePeriod": [
                                    {
                                          "startPeriod": 0,
                                          "limit": 566
                                    },
                                    {
                                          "startPeriod": 900,
                                          "limit": 448
                                    },
                                    {
                                          "startPeriod": 1800,
                                          "limit": 0
                                    }
                              ],
                              "startSchedule": "2020-10-16T14:30:00.000Z",
                              "duration": 2700
                        }
                  }
            }
      }
]
```

The charge unit and value should be in Watt as defined in the template.

LucasBrazi06, 2020-10-16:
Okay, this is a regression.
@OliveGerste Could you have a look?

OliveGerste, 2020-10-19:
@jerome-benoit @LucasBrazi06 
We push the cp in watts but it is always saved and logged in amps, because the conversion is done in the vendor implementation and the saving is done in OCPPUtils. 
So not really a bug, but a bit confusing maybe.

LucasBrazi06, 2020-10-19:
> @jerome-benoit @LucasBrazi06
> We push the cp in watts but it is always saved and logged in amps, because the conversion is done in the vendor implementation and the saving is done in OCPPUtils.
> So not really a bug, but a bit confusing maybe.

Exact! Forgot that point indeed!
@jerome-benoit You'll see the Watts in the real OCPP charging plan pushed to the charging station.
@OliveGerste Do we use the **chargingRateUnit** when we request the composite schedule?

OliveGerste, 2020-10-19:
@LucasBrazi06 Yes we use the chargingRateUnit. That is why the values from the station are so weird.
limitWatts: 651 --> this is the limit returned from the station
limit: 2.8 --> this is what we convert with the chargingRateUnit (division by 230V) (method: convertFromVendorChargingSchedule)

LucasBrazi06, 2020-10-19:
> Yes we use the chargingRateUnit. That is why the values from the station are so weird.
> limitWatts: 651 --> this is the limit returned from the station
> limit: 2.8 --> this is what we convert with the chargingRateUnit (division by 230V) (method: convertFromVendorChargingSchedule)

Then if we request the Charging Profile using Watt in getCompositeSchedule and we got Amps instead of Watts, then this is a bug :-), agree?

OliveGerste, 2020-10-19:
In my understanding yes. This is what is returned by the cs unfiltered:
[image]

LucasBrazi06, 2020-10-19:
> In my understanding yes. This is what is returned by the cs unfiltered:

Ok thanks!
@jerome-benoit Could you contact Delta and open them an issue?
Thanks!

jerome-benoit, 2020-10-19:
I take over that issue then but in asana. 
For the automation that can be done from OCPP parameter, let's discuss it later.

jerome-benoit, 2021-02-25:
Latest firmware version 3.3.6 installed on Delta charging station. 

@OliveGerste could you please check if the issue is fixed or not?

OliveGerste, 2021-02-25:
@jerome-benoit
Seems not to be fixed. It is still returned in amps per phase:
[image]

closed 2021-07-31 by LucasBrazi06