sap-labs-france/ev-server #1858: Cannot stop a running transaction as admin if the charger is offline.

jerome-benoit, 2020-10-23:


LucasBrazi06, 2020-10-23:
I think I already fixed it, maybe not deployed in prod...

jerome-benoit, 2020-10-23:
> I think I already fixed it, maybe not deployed in prod...

Ok: https://github.com/LucasBrazi06/ev-dashboard/commit/71c59e573d0f4e1ec175eeb1eea019b3ec5fa0b6 

There's no similar checks in the backend?

LucasBrazi06, 2020-10-23:
> There's no similar checks in the backend?
Nope.

jerome-benoit, 2020-10-23:
I think I've found another similar issue: I cannot delete a charger that have a non existent (or at least not visible in the dashboard) session running on testcharger tenant in QA: ABE_Electricite_SAR_Cp3 ;)

LucasBrazi06, 2020-10-23:
> I think I've found another similar issue: I cannot delete a charger that have a non existent (or at least not visible in the dashboard) session running on testcharger tenant in QA: ABE_Electricite_SAR_Cp3 ;)

If that's reproducible and can happen, yes, this has to be fixed :)

jerome-benoit, 2020-10-23:
> > I think I've found another similar issue: I cannot delete a charger that have a non existent (or at least not visible in the dashboard) session running on testcharger tenant in QA: ABE_Electricite_SAR_Cp3 ;)
> 
> If that's reproducible and can happen, yes, this has to be fixed :)

A second charger have the same issue: TUD093003B so I say it's reproducible. The culprit transaction id must be found now ...

jerome-benoit, 2020-12-16:
Need to tested to reproducibility.

closed 2021-08-05 by LucasBrazi06