sap-labs-france/ev-server #1862: Smart Charging - Handle charging stations, which refuse the charging profile

OliveGerste, 2020-10-26:


LucasBrazi06, 2020-11-03:
Retry 3 times to push the charging profiles and if failed, exclude the charging station from the Smart Charging, send a notification to the Admins and retry again to apply the Smart Charging on the Site Area.

jerome-benoit, 2020-11-03:
I think axios is used by the SetChargingProfile OCPP command, so it already retries 3 (tunable) times before throwing an error.

LucasBrazi06, 2020-11-03:
> I think axios is used by the SetChargingProfile OCPP command, so it already retries 3 (tunable) times before throwing an error.

Perfect so no need to check that part then 👍

jerome-benoit, 2020-11-03:
In fact, axios will not retry because it's not an HTTP error ;) 
The retry condition lie in the upper network layer protocol.

LucasBrazi06, 2020-11-03:
> In fact, axios will not retry because it's not an HTTP error ;)
> The retry condition lie in the upper network layer protocol.

Perfect so we need to check that part then 👍 :-)

closed 2021-08-05 by LucasBrazi06