sap-labs-france/ev-server #1886: Billing > invoices not deleted

Ninjeneer, 2020-11-05:
Local invoices are not deleted during sync, while Stripe invoices are.

ClaudeROSSI, 2021-05-24:
@Ninjeneer : Hi Loan, do you remember the purpose of that old ev-server issue (from November last year)?
Thanks

Ninjeneer, 2021-05-24:
Hi @ClaudeROSSI, if I remember well, if you delete an invoice on the Stripe side, and then run a sync on e-Mobility, the e-Mobility invoice is not deleted.
But I think it is already fixed

ClaudeROSSI, 2021-05-24:
@Ninjeneer : Thanks for the reply.

I think the issue does not match anymore the STRIPE integration as it is today. 

First of all, it is important to have in mind that invoices cannot be deleted on the stripe side when running in a productive environment. In other words, the issue is meaningless in the real context.

Anyway, the logic has been changed completely. Invoices that are visible on the eMobility side are the ones corresponding to an EV charging session (i.e.: a transaction). Invoices created manually on the stripe side (by selling some other products) are not replicated / nor visible in the eMobility database. As a consequence the deletion topic does not apply anymore.

I'll close this issue.
Let me know if you have a different view on the initial concern.

closed 2021-05-24 by ClaudeROSSI