sap-labs-france/ev-server #1888: Bug - Fix Amperage Limit

OliveGerste, 2020-11-06:
-	Update the method getChargingStationAmperage() to return the safe min/max value 
-	For handle wrong amperageLimit values : when OCPP Status Notif is received, in method enrichChargingStationConnectorWithTemplate(), you can check the amperageLimit and update with the min/max safe value

OliveGerste, 2020-11-16:
@jerome-benoit 
I think I found the part in the code where the amperageLimit is set to 0. My guess the difference is caused by different templates (or the handling of the difference):

The processing is done in the CheckChargingStationTemplateTask
There we have the following code snipped:
```
for (const connector of chargingStation.connectors) {
          // Amperage limit
          if (!Utils.objectHasProperty(connector, 'amperageLimit')) {
            connector.amperageLimit = connector.amperage;
            chargingStationUpdated = true;
          } else if (Utils.objectHasProperty(connector, 'amperageLimit') && connector.amperageLimit > connector.amperage) {
            connector.amperageLimit = connector.amperage;
            chargingStationUpdated = true;
          }
```
The template of SAP Mougins 15 does not have an amperage value on connector level right?
That iswhy the amperageLimit is set to 0 when the task runs.

For example SAP Mougins 12 has an amperage value on connector level. That is why the amperageLimit would be set this case.

LucasBrazi06, 2020-11-19:
@jerome-benoit 
Is there something else needed?

jerome-benoit, 2020-11-25:
> @jerome-benoit
> Is there something else needed?

There's no automated adjustment of the amperageLimit value if it's below the ev ready min for the charging station. I think it's checked a setting the value so it's maybe not necessary add an adjustment in the same code path.

jerome-benoit, 2020-11-25:
Done in https://github.com/LucasBrazi06/ev-server/pull/1944

closed 2020-12-02 by OliveGerste