sap-labs-france/ev-server #1891: Charging profile > Cannot clear profile with a schedule set to 0A

jerome-benoit, 2020-11-08:
```
[
      {
            "error": "Cannot set the minimum power limit to 0A, minimum expected 13A",
            "stack": "Error: Cannot set the minimum power limit to 0A, minimum expected 13A
   at AtessChargingStationVendorIntegration.setStaticPowerLimitation (/home/vcap/app/dist/webpack:/src/integration/charging-station-vendor/ChargingStationVendorIntegration.ts:64:13)
   at AtessChargingStationVendorIntegration.clearChargingProfile (/home/vcap/app/dist/webpack:/src/integration/charging-station-vendor/ChargingStationVendorIntegration.ts:285:24)
   at Function.clearAndDeleteChargingProfile (/home/vcap/app/dist/webpack:/src/server/ocpp/utils/OCPPUtils.ts:1321:7)
   at RequestMapper.handleDeleteChargingProfile (/home/vcap/app/dist/webpack:/src/server/rest/v1/service/ChargingStationService.ts:541:7)
   at CentralRestServer.restServiceSecured (/home/vcap/app/dist/webpack:/src/server/rest/CentralRestServerService.ts:336:7)"
      }
]
```

LucasBrazi06, 2020-11-08:
@OliveGerste 
Could you check if this is a dups of the one you are working on?

OliveGerste, 2020-11-09:
@LucasBrazi06 Yes it is basically the same issue.

LucasBrazi06, 2020-11-09:
Dups

closed 2020-11-09 by LucasBrazi06