sap-labs-france/ev-server #1893: Edit Site Area Bug

mahdibenromdhane, 2020-11-09:
Edit site area popup empty for basic user after create. 

[image]

OliveGerste, 2020-12-14:
@mahdibenromdhane 
It is working in prod and in local environment, but somehow not in qa: 
[image]

mahdibenromdhane, 2020-12-16:
> @mahdibenromdhane
> It is working in prod and in local environment, but somehow not in qa:
> [image]

Cannot reproduce on QA, seems to be working on my side. @OliveGerste did you make sure that the basic test-user is a site admin ?

closed 2021-08-05 by LucasBrazi06