sap-labs-france/ev-server #1921: Onboarding

stefania-santimbrean, 2020-11-23:
Ask for access

- [ ] Ask to be added to DL Charge-Angels Dream Team (External)​

- [ ] Ask for GitHub access to ev-scp-sap-it and ev-dashboard-new

stefania-santimbrean, 2020-11-23:
**Trainings to take**

Needed
- [ ] Web Progamming Foundations - 2h
- [ ] TypeScript - 1.5h
- [ ] Swagger - 2h
- [ ] REST API - 1h

Optional
- [ ] npm - 1h
- [ ] Node.js - 0.6h
- [ ] MongoDB - 0.6h
- [ ] Git - 1h

stefania-santimbrean, 2020-11-23:
**Dev Setup**

- [ ] Database
- [ ] Servers
- [ ] DB Tools
- [ ] DB Cleanup
- [ ] Register Your User
- [ ] Unit Test
- [ ] Postman
- [ ] VS Code IDE

LucasBrazi06, 2020-11-23:
> * Ask for GitHub access to ev-scp-sap-it and ev-dashboard-new

Done for ev-scp-sap-it repo.
There is no more ev-dashboard-new repo but ev-dashboard and you are already a member of it

stefania-santimbrean, 2020-11-23:
@LucasBrazi06 thanks! If I look in my Address Book I does not appear that I am a Member of DL Charge-Angels Dream Team (External)​. Could you please add me?

LucasBrazi06, 2020-11-23:
> thanks! If I look in my Address Book I does not appear that I am a Member of DL Charge-Angels Dream Team (External)​. Could you please add me?

You are in already!

closed 2020-12-08 by stefania-santimbrean