sap-labs-france/ev-server #1923: Onboarding Alfons

Fonsu, 2020-11-23:
- [x] Ask to be added to DL Charge-Angels Dream Team (External)​
- [x] Ask for GitHub access to `ev-scp-sap-it` and `ev-dashboard`

LucasBrazi06, 2020-11-24:
Done, let me know by closing the issue

Fonsu, 2020-11-26:
- [x]  VS Code IDE
- [ ]  Database
- [ ]  Servers
- [ ]  DB Tools
- [ ]  DB Cleanup
- [ ]  Register Your User
- [ ]  Unit Test
- [ ]  Postman

Fonsu, 2020-12-10:
Trainings

- [x]  SAP Application Router
- [x]  Node.js
- [ ]  MongoDB

jerome-benoit, 2020-12-18:
You are very close to be assigned your first task via GitHub issue.

jerome-benoit, 2021-01-05:
@Fonsu: I think you're now ready to start your first development task. 

Could you please confirm? 

Thanks.

Fonsu, 2021-01-06:
@jerome-benoit yes, I'm ready to start my first task

closed 2021-01-12 by LucasBrazi06