sap-labs-france/ev-server #1942: Bug in displaying Images in car Catalog carousel

mahdibenromdhane, 2020-11-25:
[image]

OliveGerste, 2020-12-21:
@mahdibenromdhane  
Image is displayed, when it is available
When no image is available, there could be a message or a standard no image icon 
(If this is not needed, it can be closed)

mahdibenromdhane, 2020-12-21:
> @mahdibenromdhane
> Image is displayed, when it is available
> When no image is available, there could be a message or a standard no image icon
> (If this is not needed, it can be closed)

Yes you'r right, I think the tab Images should not be displayed at all if there is no images to display in the caroussel. I will push a fix.

LucasBrazi06, 2020-12-21:
> Yes you'r right, I think the tab Images should not be displayed at all if there is no images to display in the caroussel. I will push a fix.

I would prefer to keep the tab with a message like in the empty list that replaces the carousel control saying that there is no car image to display.

mahdibenromdhane, 2020-12-21:
> > Yes you'r right, I think the tab Images should not be displayed at all if there is no images to display in the caroussel. I will push a fix.
> 
> I would prefer to keep the tab with a message like in the empty list that replaces the carousel control saying that there is no car image to display.

Ok.

closed 2021-08-05 by LucasBrazi06