sap-labs-france/ev-server #2002: Car Image Bug

mahdibenromdhane, 2021-01-05:


mahdibenromdhane, 2021-01-05:
[image]
[image]

LucasBrazi06, 2021-01-06:
Much better but I still have some without image.
Is it a blank one or a pb in the DB?
[image]

mahdibenromdhane, 2021-01-06:
> Much better but I still have some without image.
> Is it a blank one or a pb in the DB?
> [image]

I think it's a problem in the DB since I have it resolved in my local machine with the same cars:
[image]

LucasBrazi06, 2021-01-06:
Have you imported the latest prod DB content?
We can test in the QA also, no problem.

mahdibenromdhane, 2021-01-07:
> > Have you imported the latest prod DB content?
> > We can test in the QA also, no problem.
> 
> Sorry my bad!
> I had not synchronized your backend's PR due to a repo filtering in ZenHub.
> It works indeed perfectly!

No problem,
Great :)

WillyLellouch, 2021-01-15:
[image]
[image]

While checking this functionality I came across this. It seems weird to me that there are pictures in the Pictures tab but no picture in the main tab. Is this normal ?

mahdibenromdhane, 2021-01-15:
> [image]
> [image]
> 
> While checking this functionality I came across this. It seems weird to me that there are pictures in the Pictures tab but no picture in the main tab. Is this normal ?

It's fixed in this [issue](https://github.com/LucasBrazi06/ev-server/issues/2028).

WillyLellouch, 2021-01-15:
@mahdibenromdhane 
Well the problem still seems to exist on the test tenant at least.

mahdibenromdhane, 2021-01-15:
> @mahdibenromdhane
> Well the problem still seems to exist on the test tenant at least.

Because it's in Pending for QA deployment

WillyLellouch, 2021-01-15:
@mahdibenromdhane 
Yes sorry, I was seeing the issue in the QA column, my bad.
So anyway, concerning this issue, it is tested and works as intended.
I'll close this one

closed 2021-01-15 by WillyLellouch