sap-labs-france/ev-server #2004: mochatest

JRGONGk, 2021-01-06:
Hi!,
My ev-dashboard is connected with ev-server. But whenever I run npm run  mochatest:createContext it always show an error "  Error: connect ECONNREFUSED 127.0.0.1:8081". I have already came this far. Please help with this issue.
Thanks in advance!

[image]

rccess, 2021-08-14:
i am facing same issue, why this is closed with giving any clue , please do advise

LucasBrazi06, 2021-08-16:
Sorry guys, I'm currently on holiday but I'll try to help you.
Have you created the local.json file needed to run the unit tests in directory 'test/config/'?

rccess, 2021-08-16:
I have local.json under the test/config 
And the data under the local.json according to describe but it is getting 500 internal server error  in return

LucasBrazi06, 2021-08-23:
It seems trying to access the backend without success but managed to connect to the DB.
Have you checked the logs or try to put a breakpoint on the use case that failed?

LucasBrazi06, 2021-09-02:
Closed for inactivity reason.

closed 2021-09-02 by LucasBrazi06