sap-labs-france/ev-server #2022: Billing > STRIPE Setup - Default Values are confusing

ClaudeROSSI, 2021-01-13:
[image]

LucasBrazi06, 2021-01-13:
@ClaudeROSSI how did you end-up with your email in the Stripe URL input field?
There is no default value here.

Ninjeneer, 2021-01-14:
Indeed, there is no default value.
It seems like your browser indentified the combo [clear text field + password text field] as a login form, and autofilled it with your stored credentials.
We could eventually disable autofilling on those fields to avoid a such confusion for other users.

ClaudeROSSI, 2021-01-15:
Right. It sounds like we simply need to set autocomplete="off" on all input fields to prevent this behavior.
I'll check it.

ClaudeROSSI, 2021-01-19:
autocomplete="off" does not work (with Chrome). I tried it both on the form and on the input fields.
A lot of forums are discussing this issue ... 
I'm not yet familiar with "angular material" stuff ... I need to investigate a bit more to see if we can prevent those default values.

ClaudeROSSI, 2021-01-25:
According to Mahdi the trick here is to simply put autocomplete="new-password" 
I'll do it at least on the private key input field.

closed 2021-07-31 by LucasBrazi06