sap-labs-france/ev-server #2024: Change the crypto key in the UI -> Migration

stefania-santimbrean, 2021-01-13:
- Read: Crypto key + algo details returned from the endpoint to the UI (crypto key is hashed)
  - Update:
    - Crypto key checked if changed
    - Algo details checked (check consistency with the provide key)
    - Save Crypto key settings from the UI
      - Store the new key and the old one in a new prop: formerKey (contains the old key + key settings)
      - Set migrationToBeDone to true
      - Trigger the migration:
        - Put a lock
        - Cypher.migrate(tenantID)
          - Cypher.migrateSettings(tenantID, newKey, formerKey)
            - Load all the Settings
              - Check if this setting has already been migrated
                - Check the field formerSensitiveData
                - [low] You can double check the formerHash key
              - Check for sensitive data
                - sensitiveData = ['password', 'client_secret']
                - Create this new property
                  formerSensitiveData = {
                    password: 'formerValueEncrypted',
                    client_secret: 'formerValueEncrypted',
                    formerKeyHash: '',
                  }
                - Migrate the current sensitive data using the formerKey/newKey
                - Save the setting
        - Clean up of former sensitive data
        - The Migration is done: set migrationToBeDone to false
        - Release a lock
    - Block the update settings endpoint if the migrationToBeDone = true

LucasBrazi06, 2021-02-27:
@roxiciobotea210 
ev-server: I need more unit tests to cover all changes possible + failure handling...
ev-dashboard: I need more checks in the Crypto Settings UI

roxiciobotea210, 2021-03-02:
@LucasBrazi06 @jerome-benoit 
Could you please provide a little more details? 
- regarding ev-server changes: What do you mean by "all changes possible"? From my understanding, the current test checks the case where all values are changed. Should I also check what happens when each one is changed individually? Regarding failure handling - do you mean testing the cases for which you added new error types to make sure the outcome is the one we expect?
- regarding ev-dashboard changes: I suppose you are referring to checking the lengths/values of the fields? if so can you tell me which they are?

jerome-benoit, 2021-03-02:
> @LucasBrazi06 @jerome-benoit
> Could you please provide a little more details?
>
> * regarding ev-server changes: What do you mean by "all changes possible"? From my understanding, the current test [...]

Unit tests shall ensure that: 
+ default crypto properties are working; 
+ key + block size change works (the key length is dependent of the block size);
+ key + operation change works; 
+ key + algo change works; 
+ key + algo + block size change works;
+ key + block size + operation change works; 
+ key + algo + operation mode change works; 

Ideally, you should pick up randomly any combination from the list in Constants but that will be too long. 
Test for main: 
+ operation mode change ctr -> gcm, gcm -> ctr, ctr -> ccm, ccm -> ctr, gcm -> ccm, ccm -> gcm, ... (maybe also ofb)
+ block size change (with the key length change): 128 -> 256, 256 -> 128, 128 -> 192, 192 -> 128, ...
+ symmetric algo: aes -> camellia, camellia -> aes, ... 

You get the idea, you can permute all possible values as long the string build is part of the list in Constants file, just limit the permutation in unit tests to the more meaningful ones.

You have also to check that unsupported key properties refuse properly to run the migration.  

> * regarding ev-dashboard changes: I suppose you are referring to checking the lengths/values of the fields? if so can you tell me which they are?

key length = block size / 8

LucasBrazi06, 2021-03-02:
Thanks @jerome-benoit  !!!

jerome-benoit, 2021-03-02:
> Thank you!

Do not forget that you have to work on top of that PR: https://github.com/Charge-Angels/ev-server/pull/2139

closed 2021-08-05 by LucasBrazi06