sap-labs-france/ev-server #2031: Locking: Create exclusive lock with timeout

LucasBrazi06, 2021-01-13:
An exclusive lock should be requested with an optional timeout in ms.
When the lock cannot be acquired, a retry will be done until the timeout is reached.
If the blocking lock is released in the meantime, then the new one is set and the flow of the code can resume.

LucasBrazi06, 2021-01-13:
First customer is smart charging @OliveGerste ;-)

jerome-benoit, 2021-03-08:
What the difference with https://app.zenhub.com/workspaces/charge-angels-5dfcfa78b123df328c12f43d/issues/charge-angels/ev-dashboard/2285?

LucasBrazi06, 2021-03-08:
This issue is to try to acquire a lock for a given period of time before it fails and the other is to delete a lock automatically after a given period of time.

LucasBrazi06, 2021-03-16:
@jerome-benoit Would be nice to have unit tests.
Thanks.

closed 2021-08-05 by LucasBrazi06