sap-labs-france/ev-server #2057: Export CSV: Test and Improve performances + Check accent once imported in Excel

LucasBrazi06, 2021-01-26:
Check perfs with exporting 100k records (logs) (+= vs join()) and apply the most efficient one.

Also accents are misinterpreted in Excel:
[image]

cabch, 2021-03-01:
+ add tags export in sessions (tag description join with tag - {withTags: boolean} to add in transactionStorage

cabch, 2021-03-02:
Do i really need to test with 100k + records ? we already have 10 sec apart with 24k records !! (winner is `+=` 🥇  export lasts 260 seconds)

LucasBrazi06, 2021-03-02:
> Do i really need to test with 100k + records ? we already have 10 sec apart with 24k records !! (winner is `+=` 🥇 export lasts 260 seconds)

It's okay, anyway I limited the export to 100k records even if you requested millions ;-).
🥇 ?
So we go with the `+=` then right?

cabch, 2021-03-02:
ok, yes we go with `+=` 
(🥇 = golden medal for the `+=` ... ;))

LucasBrazi06, 2021-03-02:
> ok, yes we go with `+=`
> (🥇 = golden medal for the `+=` ... ;))

LOL
Okay :-)
So will you treat the add of the badge ID in session export in another issue?

closed 2021-03-02 by cabch