sap-labs-france/ev-server #2063: database login mongod

giorgio80, 2021-01-29:
I'm obtaining an error on central server connecting to db: 

ubuntu 20.04 

 I have created evse-user per instruction and I'm able to login via shell using: 

`mongo -u evse-user -p mypass --authenticationDatabase evse` 
 from shell with our any problem 

I then inserted config into config.json 

` "Storage": {
    "implementation": "mongodb",
    "host": "127.0.0.1",
    "port": 27017,
    "user": "evse-user",
    "password": "mypass",
    "schema": "evse"
  },`

when running 
`npm run start`

I got an error 
`Connecting to 'mongodb' in master...
{ MongoError: Authentication failed.
    at MessageStream.messageHandler (/root/ev-server/node_modules/mongodb/lib/cmap/connection.js:268:20)
    at MessageStream.emit (events.js:198:13)
    at processIncomingData (/root/ev-server/node_modules/mongodb/lib/cmap/message_stream.js:144:12)
    at MessageStream._write (/root/ev-server/node_modules/mongodb/lib/cmap/message_stream.js:42:5)
    at doWrite (_stream_writable.js:415:12)
    at writeOrBuffer (_stream_writable.js:399:5)
    at MessageStream.Writable.write (_stream_writable.js:299:11)
    at Socket.ondata (_stream_readable.js:710:20)
    at Socket.emit (events.js:198:13)
    at addChunk (_stream_readable.js:288:12)
    at readableAddChunk (_stream_readable.js:269:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)
  operationTime:
   Timestamp { _bsontype: 'Timestamp', low_: 1, high_: 1611927317 },
  ok: 0,
  code: 18,
  codeName: 'AuthenticationFailed',
  '$clusterTime':
   { clusterTime:
      Timestamp { _bsontype: 'Timestamp', low_: 1, high_: 1611927317 },
     signature: { hash: [Binary], keyId: [Long] } },
  name: 'MongoError' }
(node:4848) UnhandledPromiseRejectionWarning: Error: Not supposed to call getCollection before database start
    at MongoDBStorage.getCollection (/root/ev-server/dist/webpack:/src/storage/mongodb/MongoDBStorage.ts:33:13)
    at Function.saveLog (/root/ev-server/dist/webpack:/src/storage/mongodb/LoggingStorage.ts:82:29)
(node:4848) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:4848) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handl`

the strange things is that looking at mongoldb logs it seems that database is not defined: 

`{"t":{"$date":"2021-01-29T13:35:21.979+00:00"},"s":"I",  "c":"ACCESS",   "id":20249,   "ctx":"conn8","msg":"Authentication failed","attr":{"mechanism":"SCRAM-SHA-1","principalName":"evse-user","authenticationDatabase":"undefined","client":"127.0.0.1:45472","result":>
`

has someone else got same problem ? could u drive me in right direction ? 

Thanks 

g

giorgio80, 2021-01-29:
Solved ....

I leave here for reference to others. 

I specified "database" value in storage configuration then everything started

closed 2021-01-29 by LucasBrazi06

roiko, 2021-04-20:
I think it would be helpful to update the doc here [](https://github.com/sap-labs-france/ev-server#configuration-1)