sap-labs-france/ev-server #2084: Car Thumb images missing

mahdibenromdhane, 2021-02-08:
[image]

LucasBrazi06, 2021-02-15:
@mahdibenromdhane 
Just imported the prod DB and ran the migration tasks but still got empty thumbnails.

[image]

Could you check?

mahdibenromdhane, 2021-02-17:
> @mahdibenromdhane
> Just imported the prod DB and ran the migration tasks but still got empty thumbnails.
> 
> [image]
> 
> Could you check?

We need to trigger the sync car catalogs manually so we can apply this [PR](https://github.com/LucasBrazi06/ev-server/pull/2029) and run the migration task.

closed 2021-08-05 by LucasBrazi06