sap-labs-france/ev-server #2091: Security > Sensitive Data: Obfuscate HTTP body sensitive params in logs (POST) in addition to the existing ones

LucasBrazi06, 2021-02-09:
[image]

roxiciobotea210, 2021-02-22:
Since I recently worked with sensitive data obfuscation (writing the tests) I have a remark on this one - client_id & client_secret are already part of the fields being obfuscated. Is there some other field to add?

jerome-benoit, 2021-02-22:
You just have to ensure your tests cover that case. 
Once your PR will be merged, that issue will also be closed.

roxiciobotea210, 2021-02-22:
> You just have to ensure your tests cover that case.
> Once your PR will be merged, that issue will also be closed.

Will add specific tests for both cases, thanks!

closed 2021-08-05 by LucasBrazi06