sap-labs-france/ev-server #2099: How to connect charger?

faisalm786, 2021-02-13:
My settings are as follows

   "CentralSystems": [
    {
      "type": "ocpp",
      "implementation": "soap",
      "protocol": "http",
      "host": "localhost",
      "port": 8000,
      "debug": false
    },
    {
      "type": "ocpp",
      "implementation": "json",
      "protocol": "ws",
      "host": "localhost",
      "port": 9000,
      "keepaliveinterval": 30,
      "debug": false
    }
  ],

When I connect a charger at port 9000, with ws://localhost:9000/R77

it doesn't connect.

LucasBrazi06, 2021-02-15:
Have you checked the logs?
Moreover a connection of a charger requires a token generated in the front-end for a given tenant (organization).

faisalm786, 2021-02-17:
Hi I am not sure what the url should be to connect the charger and what port.

as far as I understood it was to connect to the backend ws://localhost:9000/[charger_address]

faisalm786, 2021-02-17:
Is there anything I am missing? Please let me know. A little noobish here.

LucasBrazi06, 2021-02-17:
> Hi I am not sure what the url should be to connect the charger and what port.
> 
> as far as I understood it was to connect to the backend ws://localhost:9000/[charger_address]

It's here:
[image]

faisalm786, 2021-02-18:
I don't see ONBOARD NEW STATION in my local setup. How to enable it?

LucasBrazi06, 2021-02-18:
> I don't see ONBOARD NEW STATION in my local setup. How to enable it?

What do you see? :)

nkazbekov, 2021-03-22:
I have the same issue and tokens are not generated on onboard new station page:
[image]

closed 2021-07-31 by LucasBrazi06