sap-labs-france/ev-server #2151: STRIPE - Setup Payment Method

ClaudeROSSI, 2021-03-05:
We need a UI allowing basic users to enter their favorite payment method. 
The User will have an additional tab where the list of existing payment methods are listed.
From that list they can press the "+" button to add a new payment method.

For some technical reasons, the UI might be different depending on which billing option is selected. The first implementation will be STRIPE specific.

**Legal constraints:**

"SCA regulation requires that you authenticate your customer up front if you intend to collect payments from them again in the future. If the customer never authenticated initially, their bank may decline future payments and ask for additional authentication."

**Priorities:**
[] Let's be pragmatic, the list itself is not mandatory. We basically need to know whether the user has a payment method or not!

**Constraints:**
[] The new tab is only visible when the billing module is activated

**Technical details (based on the POC)**

To be compliant with SCA regulation rules the flow will be one one described at:
- https://stripe.com/docs/payments/intents
additional information at:
- https://stripe.com/docs/payments/save-and-reuse

LucasBrazi06, 2021-03-05:
> * The new tab is only visible when the billing module is activated

This is also driven by the user's authorizations (canReadPayment(), canListPayments(), canUpdatePayment(), canDeletePayment(), canCreatePayment()...)

ClaudeROSSI, 2021-03-05:
I would like to avoid that we replicate payment methods on our side. This means that the UI may not provide nice filtering capabilities that we usually have! ... but here we do not expect a lot of entries. so ... this should be ok. right?

LucasBrazi06, 2021-03-05:
> I would like to avoid that we replicate payment methods on our side. This means that the UI may not provide nice filtering capabilities that we usually have! ... but here we do not expect a lot of entries. so ... this should be ok. right?

No need of filtering here indeed.

ClaudeROSSI, 2021-03-23:
We have now a merge conflict with master-qa.
The dashboard does load properly after the merge. So i'm putting back the PR to "Dev cross-review".

ClaudeROSSI, 2021-03-23:
@lucasbrazi , @jerome-benoit:
I'm also getting the same error with master-qa. Dashboard does not start anymore.

core.js:6150 ERROR TypeError: Cannot read property 'location' of undefined
    at WindowService.getHostname (window.service.ts:12)
    at WindowService.getSubdomain (window.service.ts:37)
    at new AuthenticationLoginComponent (authentication-login.component.ts:63)
    at NodeInjectorFactory.AuthenticationLoginComponent_Factory [as factory] (authentication-login.component.ts:203)
    at getNodeInjectable (core.js:3534)
    at instantiateRootComponent (core.js:10077)
    at createRootComponent (core.js:12389)
    at ComponentFactory$1.create (core.js:25072)
    at ViewContainerRef.createComponent (core.js:23112)
    at RouterOutlet.activateWith (router.js:5294)

LucasBrazi06, 2021-03-23:
> I'm also getting the same error with master-qa. Dashboard does not start anymore.

Yep, just got that in QA :-(
I just asked Jerome.

LucasBrazi06, 2021-03-23:
> > I'm also getting the same error with master-qa. Dashboard does not start anymore.
> 
> Yep, just got that in QA :-(
> I just asked Jerome.

Jerome just fixed it.

closed 2021-03-31 by ClaudeROSSI