sap-labs-france/ev-server #2158: Bugged Firmware Download

WillyLellouch, 2021-03-08:
It's now http://reverse-proxy.e-mobility-group.com:8082/api/chargingstations/firmware/<filename> for the QA.
But it seems to have issue since the API changes

jerome-benoit, 2021-03-08:
> It's now http://reverse-proxy.e-mobility-group.com:8082/api/chargingstations/firmware/ <http://reverse-proxy.e-mobility-group.com:8082/api/chargingstations/firmware/> for the QA.
> 
> 
Locally it’s http://localhost:<configuration_port>/v1/docs <http://localhost:%3Cconfiguration_port%3E/v1/docs> to access the API swagger documentation

--
Jérôme Benoit aka fraggle
Piment Noir - https://piment-noir.org
OpenPGP Key ID : 27B535D3
Key fingerprint : B799 BBF6 8EC8 911B B8D7 CDBC C3B1 92C6 27B5 35D3

WillyLellouch, 2021-03-15:
This now works properly.
Tested with remote firmware update with Joint-Lighting Station
Closing Issue

closed 2021-03-15 by WillyLellouch