sap-labs-france/ev-server #2168: Company > Authorizations: Provide unit tests for check company auths including the site admin role

LucasBrazi06, 2021-03-10:
List the possible unit tests on company in this issue and implement them

roxiciobotea210, 2021-03-10:
Should the unit tests be for CRUD operations? Or more like testing with different roles to see the correct results are returned?

LucasBrazi06, 2021-03-10:
> Should the unit tests be for CRUD operations? Or more like testing with different roles to see the correct results are returned?

For CRUD & different roles :).
there are already some of them I think.
The goal is to test all the combinations against the backend to ensure there are no security breaches and avoid regressions.

roxiciobotea210, 2021-03-10:
> For CRUD & different roles :).
> there are already some of them I think.
> The goal is to test all the combinations against the backend to ensure there are no security breaches and avoid regressions.

I found some tests for Admin role in `CompanyOrgTest.ts` file. Shall we complete the file with tests for the other roles or create a new test file like `AuthrizationTest.ts`?

LucasBrazi06, 2021-03-10:
> I found some tests for Admin role in `CompanyOrgTest.ts` file. Shall we complete the file with tests for the other roles or create a new test file like `AuthrizationTest.ts`?

You can reuse the same file

roxiciobotea210, 2021-03-11:
For admin user the following tests exist:
- [ ] Check that an admin user is able to create a company
- [ ] Check that an admin user is able to retrieve the created company
- [ ] Check that an admin user is able to find the company in the company list
- [ ] Check that an admin user is able to update a company
- [ ] Check that an admin user is able to read the updated company
- [ ] Check that an admin user is able to delete a company
- [ ] Check that an admin user is not able to find the company after it was deleted 

For basic user we shall implement:
- [ ] Check that a basic user is able to read a company if he is assigned to a site of that company
- [ ] Check that a basic user is not able to read a company  if he is not assigned to a site of that company
- [ ] Check that a basic user is not able to read a company after he was removed from the site assigned to that company
- [ ] Check that a basic user is not able to create a company
- [ ] Check that a basic user is not able to update a company
- [ ] Check that a basic user is not able to delete a company

roxiciobotea210, 2021-03-11:
@LucasBrazi06 Since only admin user is able to create, delete, update Company, I think the tests added for basic user are the same for all other roles, right?
Should we still add the same tests for all other roles or is it enough to have them for basic role?

LucasBrazi06, 2021-03-11:
> @LucasBrazi06 Since only admin user is able to create, delete, update Company, I think the tests added for basic user are the same for all other roles, right?
> Should we still add the same tests for all other roles or is it enough to have them for basic role?

That's ok with Basic user, we rely now on authorizations and not roles so it should be okay for all users.
Thanks Roxana!

roxiciobotea210, 2021-03-11:
Great! I created the PR.

closed 2021-03-11 by LucasBrazi06