sap-labs-france/ev-server #2183: Billing Settings - STRIPE - Make it safe when settings are not yet set

ClaudeROSSI, 2021-03-15:
When activated, and not yet properly configured, the Billing Service is throwing exception on each attempt to connect to STRIPE.

We may consider the complete billing feature as disabled when the corresponding settings are not consistent.
For instance, when the secret_key to access the STRIPE customer account is not yet set.

ClaudeROSSI, 2021-03-15:
@LucasBrazi06 : Billing is already ON on the QA tenant but with no settings (i.e.: no secret_key)
I need it to check the impact of the wrong settings.

I'm not sure how to proceed client side when checking whether the BILLING component is active or not.
... if (this.componentService.isActive(TenantComponents.BILLING) 

Is this isActive()  supposed to return false when the settings are not consistent?
Thanks for your help,

ClaudeROSSI, 2021-03-15:
Actually we may consider it as a Bug because the user can see the "Invoice" module while the Billing settings are not yet set!

LucasBrazi06, 2021-03-15:
> Actually we may consider it as a Bug because the user can see the "Invoice" module while the Billing settings are not yet set!

Like Concur, the user has access to the feature but he got a relevant error message when this does not work.
So we do not consider that the settings are valid to activate or not the UI components.

ClaudeROSSI, 2021-03-15:
> Like Concur, the user has access to the feature but he got a relevant error message when this does not work.
> So we do not consider that the settings are valid to activate or not the UI components.

ok. but throwing exceptions on StartTransaction and StopTransaction is a bit confusing for the calling layer. right?
Or do we assume ... no charging sessions should occur when the billing is not properly set?

LucasBrazi06, 2021-03-15:
> ok. but throwing exceptions on StartTransaction and StopTransaction is a bit confusing for the calling layer. right?
> Or do we assume ... no charging sessions should occur when the billing is not properly set?

We must differentiate between a technical failure due to misconfiguration from a business failure due to not having a credit card available.
For technical failure, I would say that we do not prevent the session to run.

ClaudeROSSI, 2021-03-15:
Let's try at least to avoid ugly exceptions when the secret_key is not set!!!

Examples:
[image]

closed 2021-03-16 by LucasBrazi06