sap-labs-france/ev-server #2187: Billing - STRIPE - Send Receipt to user when a payment succeeds

ClaudeROSSI, 2021-03-15:
How do we handle receipt on our side?

Do we need to persists the PDF document and send it to the customer?
or do we simply rely on STRIPE capabilities - like "Send Receipt"

[image]

LucasBrazi06, 2021-03-15:
> Do we need to persists the PDF document and send it to the customer?
> or do we simply rely on STRIPE capabilities - like "Send Receipt"

Good question.
Think about another payment implementation and choose the best solution :-)

ClaudeROSSI, 2021-05-20:
I had a discussion with Hanno regarding the need of the receipt as a PDF document.
With agreed that this is not a document which is mandatory. Professional EV drivers can get reimbursed by providing the invoice only.

Moreover, STRIPE does not provide any API to retrieve a stable URL to the receipt. We therefore cannot provide the PDF using the same approach than the one we use for the invoice PDF document.