sap-labs-france/ev-server #2188: Billing - STRIPE - Payment Method - Collect Permission to Reuse Cards

ClaudeROSSI, 2021-03-15:
Get permission to save a card by informing the user how we intend to use/reuse the card for paying EV charging sessions.

ClaudeROSSI, 2021-03-15:
Recommendations from STRIPE can be found at:
- https://stripe.com/docs/strong-customer-authentication/faqs#see-also

**How do I collect permission to reuse cards?** 

When you set up your payment flow to properly save a card with the Payment Intents or Setup Intents API, Stripe marks any subsequent off-session payment as a merchant-initiated transaction (MIT) to reduce the need to authenticate. Merchant-initiated transactions require an agreement (also known as a mandate) between you and your customer. Add terms to your website or application on how you plan to process payments that your customer can opt into.

At a minimum, ensure that your terms cover the following:

The customer’s permission to you initiating a payment or a series of payments on their behalf
The anticipated frequency of payments (i.e., one-time or recurring)
How the payment amount will be determined
Add text in your checkout flow that references the terms of the payment, for example:

I authorize [your business name] to send instructions to the financial institution that issued my card to take payments from my card account in accordance with the terms of my agreement with you.

cabch, 2021-04-02:
checkbox has been added, shouldn't we save it to db as the eula acceptance is ?
shouldn't we add some legal text somewhere just as eula ?

LucasBrazi06, 2021-04-02:
> checkbox has been added, shouldn't we save it to db as the eula acceptance is ?
> shouldn't we add some legal text somewhere just as eula ?

Normally it's just a reminder of what should be already in the EULA.
Double check that our main EULA is up to date with the Billing (ev-server/src/assets/eula)

cabch, 2021-04-02:
> > checkbox has been added, shouldn't we save it to db as the eula acceptance is ?
> > shouldn't we add some legal text somewhere just as eula ?
> 
> Normally it's just a reminder of what should be already in the EULA.
> Double check that our main EULA is up to date with the Billing (ev-server/src/assets/eula)

it says 'Stripe: used for the billing and the payment of the sessions'

It's not clearly told that bank details will be kept for future use / that 'instructions can be send to the financial institution that issued [my] card to take payments from my card account'

LucasBrazi06, 2021-04-02:
You can update the EULA then.

ClaudeROSSI, 2021-04-02:
> You can update the EULA then.

We can update the EULA text but this seems not to be sufficient. Especially for reusing card information for future payments.
According to the European Data protection rules it sounds like an explicit consent from the customer to reuse the card information is mandatory.

LucasBrazi06, 2021-04-02:
> We can update the EULA text but this seems not to be sufficient. Especially for reusing card information for future payments.
> According to the European Data protection rules it sounds like an explicit consent from the customer to reuse the card information is mandatory.
> 
> do we have a contact for such legal text?

The question here is whether we need to store the user's consent or not for the payment with a timestamp and the text on which the user agreed in the DB along the credit card.
The storage of the consent, versioning of the text... is done already in the EULA.
If yes, you have to do the same approach than in the EULA.
No need a contact. Just follow Stripe advice.

ClaudeROSSI, 2021-04-06:
> The question here is whether we need to store the user's consent or not for the payment with a timestamp and the text on which the user agreed in the DB along the credit card.
> The storage of the [...]

As already mentioned, STRIPE recommendations are quite simple:
You have it at: https://stripe.com/docs/strong-customer-authentication/faqs#mandates

The pb that I see with the EULA is that this is SAP oriented. Do we plan to have a different EULA per CPO?

LucasBrazi06, 2021-04-06:
> The pb that I see with the EULA is that this is SAP oriented. Do we plan to have a different EULA per CPO?

No different EULA per CPO planned.
We can think about a user consent framework per tenant if necessary.

closed 2021-05-20 by ClaudeROSSI