sap-labs-france/ev-server #2197: Billing - STRIPE - Payment method - Billing_details

ClaudeROSSI, 2021-03-16:
payment failure events (from STRIPE) dumps additional information like the billing_details:
Let's try to populate this information when creating the payment method.
I'm not sure if this should be done server-side (or client-side)

[image]

LucasBrazi06, 2021-03-16:
> payment failure events (from STRIPE) dumps additional information like the billing_details:
> Let's try to populate this information when creating the payment method.
> I'm not sure if this should be done server-side (or client-side).
> As of today, the name shown below as been set client side !

Is this related to the user created in Stripe by e-Mobility?
In that case you should provide the user's address in e-Mobility to Stripe (even make it mandatory to the users when the billing is active.)
And when you create a payment method and use it, I would expect the user's address to be provided in the billing_details property, right?

ClaudeROSSI, 2021-03-16:
> Is this related to the user created in Stripe by e-Mobility?
> In that case you should provide the user's address in e-Mobility to Stripe (even make it mandatory to the users when the billing is [...]

needs some investigation ... sounds like stripe gets the information from the "payment_method" itself, not from the user?!??!?

ClaudeROSSI, 2021-05-06:
FYI: The country information in stripe has to be a two-character string: FR, US, etc ...

When a string such as 'France' is used some of their endpoints trigger exceptions.

We need to find a way to convert the country string to a consistent country code ...
Strange enough but a wrong value is not refused when populating the customer address!!! - to be clarified! 

[image]

ClaudeROSSI, 2021-05-24:
This issue has been solved backend side when synchronizing user/customer data and when creating a new payment method.

c.f.: PR: https://github.com/sap-labs-france/ev-server/issues/2466

I'm closing this one

closed 2021-05-24 by ClaudeROSSI