sap-labs-france/ev-server #2220: Session > Inactivity: Inactivity is too high

LucasBrazi06, 2021-03-23:
ChargeX reported a high inactivity, could you please check?
[image]

jerome-benoit, 2021-03-23:
Could be a duplicate of https://github.com/sap-labs-france/ev-dashboard/issues/2307. That should be checked.

LucasBrazi06, 2021-03-26:
@WillyLellouch Any news on this one?
Thanks.

LucasBrazi06, 2021-03-30:
@jerome-benoit Issue without PR?

LucasBrazi06, 2021-04-12:
@WillyLellouch @jerome-benoit Any news on this one?

WillyLellouch, 2021-04-19:
@LucasBrazi06 I will need the full copy of the database entries for the consumptions of this transaction. There is not enough info in the logs to determine the source of the issue.

LucasBrazi06, 2021-04-19:
> @LucasBrazi06 I will need the full copy of the database entries for the consumptions of this transaction. There is not enough info in the logs to determine the source of the issue.

You don't have the logs but you have the metervalues and consumptions collections, that should be enough.
Logs are too big to be backed up and made the dump failing.

LucasBrazi06, 2021-04-19:
> @LucasBrazi06 I will need the full copy of the database entries for the consumptions of this transaction. There is not enough info in the logs to determine the source of the issue.

You can also connect to the customer's tenant to get the logs.

WillyLellouch, 2021-04-19:
@LucasBrazi06 Yes I already have the logs from the customer's tenant. They are not enough to determine the source of the issue. And I do not have access to the database. I need to be able to see the inactivity evolve over time in the consumptions, which is not logged except in the status updates (which is insufficient)

LucasBrazi06, 2021-04-19:
> @LucasBrazi06 Yes I already have the logs from the customer's tenant. They are not enough to determine the source of the issue. And I do not have access to the database. I need to be able to see the [...]

I pushed a backup of the AWS DB in our sharepoint, I told Jerome already.
I don't get what is missing?

WillyLellouch, 2021-04-19:
@LucasBrazi06 I wasn't made aware that the backup was done. Since it is, I have everything I need. Thanks.

jerome-benoit, 2021-04-19:
> @LucasBrazi06 I wasn't made aware that the backup was done. Since it is, I have everything I need. Thanks.

I've forwarded to you the mail with the link to the export previous Thursday.

jerome-benoit, 2021-04-19:
> @LucasBrazi06 <https://github.com/LucasBrazi06> Yes I already have the logs from the customer's tenant. They are not enough to determine the source of the issue. And I do not have access to the [...]

Nothing is missing. Willy, could you please proceed with the import of the DB to dig the data and identify meter values records correlated to a ongoing transaction that might lead to incorrect inactivity computation?

Thanks.

--
Jérôme Benoit aka fraggle
Piment Noir - https://piment-noir.org <https://piment-noir.org/>
OpenPGP Key ID : 27B535D3
Key fingerprint : B799 BBF6 8EC8 911B B8D7 CDBC C3B1 92C6 27B5 35D3

WillyLellouch, 2021-04-22:
The issue is not currently reproducible. We are investigating with the simulator but it would be beneficial if we could have a dedicated station from ChargeX as it seems this particular issue is only encountered on their tenant.

LucasBrazi06, 2021-04-22:
> The issue is not currently reproducible. We are investigating with the simulator but it would be beneficial if we could have a dedicated station from ChargeX as it seems this particular issue is only encountered on their tenant.

@WillyLellouch You can request them to run again a transaction on the same charger.
Let me know.

WillyLellouch, 2021-04-23:
> @WillyLellouch You can request them to run again a transaction on the same charger.
Let me know.

This doesn't seem to be tied to any one charging station and the charging stations that have already had the issue do not necessarily have the issue again. Some have had the issue only once, other twice. There are also no commonalities in the transactions affected, be it duration, user, or any other parameter. That's one of the main issue in identifying the problem.

closed 2021-07-31 by LucasBrazi06