sap-labs-france/ev-server #2242: Eichrecht : Signed Data is not correctly transmited with some systems

WillyLellouch, 2021-03-29:
Signed Data is only taken from MeterData where it should be taken from StopTransaction as well.

jerome-benoit, 2021-03-30:
@WillyLellouch could you please give us the full technical details on the issue or make a draft PR from your fix branch for initial reviewal and consider that issue as a top priority? 

Thanks.

WillyLellouch, 2021-03-31:
@jerome-benoit draft PR coming tommorow at the latest

jerome-benoit, 2021-04-02:
> @jerome-benoit draft PR coming tomorrow at the latest

@WillyLellouch We are tomorrow since yesterday. Could you please give us a status?

WillyLellouch, 2021-04-02:
@jerome-benoit I was a bit too optimistic with the timing. I've encountered a bug in my implementation that I'm resolving today. Also, should I include my unit tests in the PR or make a separate one ?

jerome-benoit, 2021-04-02:
You can put everything in the same PR. Thanks.

jerome-benoit, 2021-04-07:
@WillyLellouch: 5 more days since without any status update on a customer critical issue. What the hell is happening?

WillyLellouch, 2021-04-08:
@jerome-benoit Sorry for the lack of communication. I've finished to implement my solution but I have some trouble with the unit tests.
As in I can't run them and I don't know why that is.
The solution I implemented is the simplest I could do. the reason it took longer than expected is that I originally went in a completely different direction that required me to rewrite large sections of code and I got lost in the weeds. My new code only changes 1 file.
Since I could not do the Unit tests properly, I'll do the PR anyway and you can see if this is satisfactory. Who must I ask for indications as to why my Unit Tests are not operative ?

jerome-benoit, 2021-04-08:
The checks at PRs do a unit tests run.

For developers support, we  have a teams and a mailing list to ask for it.

Thanks.

WillyLellouch, 2021-04-20:
The fix is non functional with ChargeX.
I'm investigating to see what went wrong in testing

WillyLellouch, 2021-04-23:
@LucasBrazi06 The tests were a success. There were no additional issues to report

LucasBrazi06, 2021-04-23:
> @LucasBrazi06 The tests were a success. There were no additional issues to report

Ok, perfect!
Thanks!

closed 2021-08-05 by LucasBrazi06