sap-labs-france/ev-server #2279: Non-admin users can read every charging stations

Ninjeneer, 2021-04-07:
Non-admin users should only be able to read charging stations of assigned sites 
[image]

But can actually read charging stations of any sites.

LucasBrazi06, 2021-04-07:
I just checked and it is working.
Could you give me more info?
Thanks.

Ninjeneer, 2021-04-07:
Abdul from ChargeX reported the issue, and I was able to reproduce it on my side.
I created a basic user on my local slf, only assigned to Mougins. I am able to read charging stations from Caen, triggering the /api/chargingstations/the_chargingstation_id

Also, it seems that there are no site checks on the handleGetChargingStation method.

LucasBrazi06, 2021-04-07:
Ah ok, because the list of charging stations is correct on my local machine.
@roxiciobotea210 Could you have a look at this issue?
Thanks

roxiciobotea210, 2021-04-07:
Hi,
I am not able to reproduce the issue yet. Just a question to you @LucasBrazi06 - when refactoring authorization for Company/Site/SiteArea was I supposed to NOT modify the `AuthorizationsDefinition.ts` file? Because I did (removed some of the custom filtering) and I'm thinking that could be causing this issue?

LucasBrazi06, 2021-04-07:
> Hi,
> I am not able to reproduce the issue yet. Just a question to you @LucasBrazi06 - when refactoring authorization for Company/Site/SiteArea was I supposed to NOT modify the `AuthorizationsDefinition.ts` file? Because I did (removed some of the custom filtering) and I'm thinking that could be causing this issue?
> 
> Edit - However I did not remove anything from the ChargingStations...
> 
> Edit 2 - I checked, it's not because of that. I will implement the check.

Then it's a bug not visible in the UI as the list is correctly filtered.
Thanks for the fix.

closed 2021-08-05 by LucasBrazi06