sap-labs-france/ev-server #2288: AUTH > Super tenant user registration broken

Ninjeneer, 2021-04-08:
[image]

Ninjeneer, 2021-04-08:
Fixed during issue creation :)

jerome-benoit, 2021-04-08:
There is still an issue: tenant with empty subdomain are allowed and it makes one unit test fails. It must not be allowed.

Ninjeneer, 2021-04-08:
Ok, I can handle it.

Just one thing : 
Currently tenant creation return a `HTTPError.USER_EMAIL_ALREADY_EXIST_ERROR` if name or subdomain are already used. It needs to be obviously changed, but is there a preferred custom HTTP code I should use, or can I choose any unused 5XX code ?

LucasBrazi06, 2021-04-08:
FYI, I already fixed the empty sub-domain in the Json schema, sent you a comment already this afternoon.

LucasBrazi06, 2021-04-08:
> Currently tenant creation returns a `HTTPError.USER_EMAIL_ALREADY_EXIST_ERROR` if name or subdomain are already used. It needs to be obviously changed, but is there a preferred custom HTTP code I should use, or can I choose any unused 5XX code ?

For the time being choose one available not yet used.
We will one day change this and always return an application error code in the response.

jerome-benoit, 2021-04-08:
I guess that issue is solved but a proper return code will be chosen, right?

LucasBrazi06, 2021-04-08:
> I guess that issue is solved but a proper return code will be chosen, right?

Yes

LucasBrazi06, 2021-04-08:
@Ninjeneer I just tested and it seems that a generic message is displayed despite the right error code is returned:

[image]

Could you check?

Ninjeneer, 2021-04-09:
Indeed, added specific errors for name/subdomains.

closed 2021-08-05 by LucasBrazi06