sap-labs-france/ev-server #2305: Reset firmware update status to "Installed" at receiving successful boot notification

jerome-benoit, 2021-04-10:


jerome-benoit, 2021-04-19:
The issue here is that the status is expected be sent after the charging station reboot to the new firmware version in the OCPP spec.

LucasBrazi06, 2021-04-19:
> The issue here is that the status is expected be sent after the charging station reboot to the new firmware version in the OCPP spec.

In other words, that's a bug on their side right?

jerome-benoit, 2021-04-19:
> > The issue here is that the status is expected be sent after the charging station reboot to the new firmware version in the OCPP spec.
> 
> In other words, that's a bug on their side right?

In any case that issue is meant to workaround bugs in the firmware installation status sent by charging station. 

But I can't add that workaround at boot notification: the firmware installation status with status "Installed" must be sent after the reboot, so after the boot notification sending and acceptation. 

It can be a scheduler task checking the firmware installation status, the charging station online status doing reset once a day.

closed 2021-07-31 by LucasBrazi06