sap-labs-france/ev-server #2333: Users CSV: users import / export

cabch, 2021-04-15:
- [ ] handle csv double quoted
- [ ] Raise error if more columns than expected during split

cabch, 2021-04-19:
@LucasBrazi06 did you mean something specific to be done for the "raise error" part ? more than what's already done ? because for now when the file contains more column than expected we already have an error like "provided file invalid"

LucasBrazi06, 2021-04-19:
No, extra columns should be ignored, only required ones trigger errors.

LucasBrazi06, 2021-04-19:
What I meant is that if you have 4 columns in the headers and you get 5 in a given line, you reject it but only the faulty line and you continue to process the rest of the users.

cabch, 2021-05-24:
Why it's iceboxed : 
Check if we do something with the withTag flag ?
Do we keep it as flag and add an option to the UI ? as flag is always set to true for now ...

closed 2021-07-31 by LucasBrazi06