sap-labs-france/ev-server #2479: Smart Charging > Ensure safety, fix known issues

OliveGerste, 2021-05-27:
- [ ] Locking Issue -> implement backup mechanism if smart charging fails
- [ ] Reduce only one phase of the root fuse, when single phased station is excluded from smart charging on three phased site area
- [ ] When excluding dc stations, take into account efficiency of station
- [ ] Charging Stations without charge point cause an error (Can happen when the charging station is not excluded from smart charging)
- [ ] If there is a problem with an active transaction (which is not existing) smart charging fails for the site area
- [ ] Unit tests to be added -> Car Management, Efficiency

jerome-benoit, 2021-05-27:
For the locking, you have to extend the current primitive with an acquire timeout and add a release time deadline.

OliveGerste, 2021-05-27:
> For the locking, you have to extend the current primitive with an acquire timeout and add a release time deadline.

I think there is already an issue for this right? https://github.com/sap-labs-france/ev-dashboard/issues/2285

The aim of the first point is to have a backup mechanism in general if smart charging fails for whatever reason.

jerome-benoit, 2021-05-27:
The backup mechanism must rely on that locking primitive. It's the easiest way to have it.

OliveGerste, 2021-05-27:
> The backup mechanism must rely on that locking primitive. It's the easiest way to have it.

The backup mechanism I have in mind would apply secure TxDefaultProfiles to each station, when smart charging is activated for a site area.
Then the algorithm can apply TxProfiles, which are used for the transactions and are able to charge more.
If smart charging fails (Maybe for another reason than locking also), it will charge the default value and we are save.

Nevertheless the locking issue needs to be solved you are right.

OliveGerste, 2021-05-28:
@jerome-benoit 
To solve the locking issue. What would be the approach. 
Do we create a scheduled task which checks if the locks are timed out.
Or do we add this functionality to the acquire methodology and if an new lock is needed to we check the old one in place is still valid? And if not releasing it.

jerome-benoit, 2021-05-28:
@OliveGerste: I propose that we do a meeting next week with @LucasBrazi06 to discuss it, along with your proposal for posting asset consumptions.

LucasBrazi06, 2021-05-28:
> @jerome-benoit
> To solve the locking issue. What would be the approach.
> Do we create a scheduled task which checks if the locks are timed out?
> Or do we add this functionality to the acquire methodology and if an new lock is needed, we check the old one in place is still valid? And if not releasing it.

Both techniques :-)
Let's discuss it during your meeting.

OliveGerste, 2021-06-22:
@LucasBrazi06 Applying the calculated charging profiles to all stations very frequently takes a lot of time. 
What do you think about checking if the charging profile changed, before we apply it again. (We already retrieve the existing ones from the database)
And we only apply it, when the profile schedule and the start changed.

LucasBrazi06, 2021-06-22:
> @LucasBrazi06 Applying the calculated charging profiles to all stations very frequently takes a lot of time.
> What do you think about checking if the charging profile changed, before we apply it [...]

Very good idea :-) 👍  !!!

closed 2021-08-05 by LucasBrazi06