sap-labs-france/ev-server #2494: billing - session history - show invoice status

ClaudeROSSI, 2021-05-31:
The session history was supposed to show (for each session) the corresponding invoice number. This can be easily implemented when the 'immediate billing' option is ON. However, with the 'periodic billing' is selected, the invoice number is not known until the end of the month. It therefore makes it difficult to maintain that information in each session row.

**Proposal**: 
Knowing that we may have a single invoice for several sessions. It would make sense to only show the billing status instead of the invoice number. 

[image]

Billing status could be: not billed, draft, billed, error or pending
- **PENDING** will appear only when the billing process is ongoing (i.e.: when an async task is about to run and change the status)
- '**draft**' denotes the fact that the billing will be completed at the end of the month - only then the invoice number will be known.
- '**not billed**' - The billing of charging session is not yet activated

The fact that we only show the status will improve the performance as well (for both the session history tab and the session in error tab). Because the information can be fetched easily and we do not need to replicate too much data (from the invoice to the session collection).

LucasBrazi06, 2021-05-31:
After the first transaction, we should create a billing item and an invoice that we keep as open right?
Why don't we have an invoice number here then?

Nevertheless, we can keep the invoice number empty, it not a problem for me.
The status of the invoice is displayed in the invoice list, no need to repeat it for each session in the history.

ClaudeROSSI, 2021-05-31:
> After the first transaction, we should create a billing item and an invoice that we keep as open right?
> Why don't we have an invoice number here then?
>
> Nevertheless, we can keep the invoice number [...]

After the first transaction the invoice is not 'open' ... it remains in a 'draft' state. This is why we do not know the invoice number. For some legal constraints (to avoid gaps in the invoice number sequence), the final invoice number is generated only when finalising the invoice (at the end of the month). 

From an end-user perspective, I think showing the status is sufficient and clear enough.

LucasBrazi06, 2021-05-31:
This does not shock me to keep only the invoice ID.
We can keep the Invoice ID with a jump to the invoice list: when not present, transaction is not billed.
We need an Invoice ID filter.

ClaudeROSSI, 2021-05-31:
> This does not shock me to keep only the invoice ID.
> We can keep the Invoice ID with a jump to the invoice list: when not present, transaction is not billed.
> We need an Invoice ID filter.

ok. But when you say ID you are talking about our invoice ID. Right? not the invoice number!?!
Do we really want to show our internal ID to the end user?

LucasBrazi06, 2021-05-31:
> ok. But when you say ID you are talking about our invoice ID. Right? not the invoice number!?!
> Do we really want to show our internal ID to the end user?

No the real invoice ID not our internal one.

ClaudeROSSI, 2021-05-31:
well ... i'm confused ;)

we have: 
- Invoice ID
- Stripe Invoice ID
- Stripe Invoice Number

We also have:
- Stripe Invoice Item ID

I would expect that the end-user only see the legal stuff: the invoice number.
But then we only have it at the end of the month.

LucasBrazi06, 2021-05-31:
Perfect with the invoice number.
Then it's empty until we have the invoice at the end of the month.
I'm ok with that except if you see any other issues?

ClaudeROSSI, 2021-05-31:
> Perfect with the invoice number.
> Then it's empty until we have the invoice at the end of the month.
> I'm ok with that except if you see any other issues?

ok.  Perfect.
Then I have to fix a bug. Because right now we do not see it event when 'Immediate billing' is on!

LucasBrazi06, 2021-05-31:
> we have:
> 
> * Invoice ID
> * Stripe Invoice ID
> * Stripe Invoice Number
> 
> We also have:
> 
> * Stripe Invoice Item ID

Btw, I don't want any 'stripe' naming in the properties :-)

ClaudeROSSI, 2021-05-31:
Well .. to be able to display the invoice Number (and not the invoice ID( I had to fix both the backend and the frontend side.
Thanks for reviewing!

closed 2021-07-26 by ClaudeROSSI