sap-labs-france/ev-server #2502: Smart Charging - fix backup asset calculation and add parameter of asset consumption

OliveGerste, 2021-06-01:
The current smart charging integration assumes that the current asset consumption is valid for 2 minutes. 
IoThink consumptions can take up to 5 minutes until they are refreshed. So it uses the high static value too often.

Goal is to add a parameter to the smart charging settings how long asset consumptions are valid. 

@LucasBrazi06 My proposal would be to add this directly in the smart charging integration settings.

Option two would be to store the value per asset, which allows you to differ validity time between different assets.

LucasBrazi06, 2021-06-01:
> @LucasBrazi06 My proposal would be to add this directly in the smart charging integration settings.
> 
> Option two would be to store the value per asset, which allows you to differ validity time between different assets.

This looks like the time interval that we have in asset to gather the data right?
Why not using this property to get the latest asset's consumption in the Smart Charging?

OliveGerste, 2021-06-02:
> This looks like the time interval that we have in asset to gather the data right?
> Why not using this property to get the latest asset's consumption in the Smart Charging?

I also thought about using the refresh time to identify if the asset consumption is expired.
But there is one problem with this:

- IoThink theoretically provides a consumption every minute, but sometimes the refresh takes e.g. 3 minutes (3 new consumptions)
- So if we take the refresh interval from the asset connector to define the expiration of an asset consumption (1 minute for ioThink), the asset consumption is only valid for one minute
- If we increase the value to 5 minutes, the refresh is also done every 5 minutes, even if after 2 minutes new consumptions are available 

If this is no problem, the solution will work. 
But then we have to increase the refresh interval of ioThink to at least 5 minutes I think.

LucasBrazi06, 2021-06-02:
To be honest ioThink should refresh their value every minute.
This is a bug on their side and they must fix it!
We can also always use the refresh interval to get the latest consumption if this is not the case today.
We default to the static value as we do today to be on the safe side.

closed 2021-07-31 by LucasBrazi06